OpenTracksApp/OpenTracks #1190: java.lang.ClassCastException: java.lang.String cannot be cast to java.util.ArrayList

brianjmurrell, 2022-04-18:
Seeing a new exception.  I think it's happening when I try to export a track to a GPX file.  I am not getting any results from trying to export in any case which is what sent me looking when I found this:

```
04-18 17:22:30.383  2597  2597 D ShareContentProvider: Created uri content://de.dennisguse.opentracks.debug.content/tracks/gpx/43/2022-04-18T16_26-04.gpx 
with MIME application/gpx+xml
04-18 17:22:30.384  2597  2597 W Bundle  : Key android.intent.extra.TEXT expected ArrayList<CharSequence> but value was a java.lang.String.  The default v
alue <null> was returned.
04-18 17:22:30.384  2597  2597 W Bundle  : Attempt to cast generated internal exception:
04-18 17:22:30.384  2597  2597 W Bundle  : java.lang.ClassCastException: java.lang.String cannot be cast to java.util.ArrayList
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.os.BaseBundle.getCharSequenceArrayList(BaseBundle.java:1343)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.os.Bundle.getCharSequenceArrayList(Bundle.java:1145)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.content.Intent.getCharSequenceArrayListExtra(Intent.java:8364)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.content.Intent.migrateExtraStreamToClipData(Intent.java:11345)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.content.Intent.migrateExtraStreamToClipData(Intent.java:11308)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.app.Instrumentation.execStartActivity(Instrumentation.java:1721)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.app.Activity.startActivityForResult(Activity.java:5314)
04-18 17:22:30.384  2597  2597 W Bundle  :      at androidx.activity.ComponentActivity.startActivityForResult(ComponentActivity.java:597)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.app.Activity.startActivityForResult(Activity.java:5272)
04-18 17:22:30.384  2597  2597 W Bundle  :      at androidx.activity.ComponentActivity.startActivityForResult(ComponentActivity.java:583)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.app.Activity.startActivity(Activity.java:5658)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.app.Activity.startActivity(Activity.java:5611)
04-18 17:22:30.384  2597  2597 W Bundle  :      at de.dennisguse.opentracks.TrackRecordedActivity.onOptionsItemSelected(TrackRecordedActivity.java:174)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.app.Activity.onMenuItemSelected(Activity.java:4269)
04-18 17:22:30.384  2597  2597 W Bundle  :      at androidx.fragment.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:352)
04-18 17:22:30.384  2597  2597 W Bundle  :      at androidx.appcompat.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:264)
04-18 17:22:30.384  2597  2597 W Bundle  :      at androidx.appcompat.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
04-18 17:22:30.384  2597  2597 W Bundle  :      at androidx.appcompat.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:66)
04-18 17:22:30.384  2597  2597 W Bundle  :      at androidx.appcompat.widget.Toolbar$1.onMenuItemClick(Toolbar.java:221)
04-18 17:22:30.384  2597  2597 W Bundle  :      at androidx.appcompat.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:781)
04-18 17:22:30.384  2597  2597 W Bundle  :      at androidx.appcompat.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:834)
04-18 17:22:30.384  2597  2597 W Bundle  :      at androidx.appcompat.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:158)
04-18 17:22:30.384  2597  2597 W Bundle  :      at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:985)
04-18 17:22:30.384  2597  2597 W Bundle  :      at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:975)
04-18 17:22:30.384  2597  2597 W Bundle  :      at androidx.appcompat.widget.ActionMenuView.invokeItem(ActionMenuView.java:625)
04-18 17:22:30.384  2597  2597 W Bundle  :      at androidx.appcompat.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:151)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.view.View.performClick(View.java:7448)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.view.View.performClickInternal(View.java:7425)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.view.View.access$3600(View.java:810)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.view.View$PerformClick.run(View.java:28305)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.os.Handler.handleCallback(Handler.java:938)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.os.Handler.dispatchMessage(Handler.java:99)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.os.Looper.loop(Looper.java:223)
04-18 17:22:30.384  2597  2597 W Bundle  :      at android.app.ActivityThread.main(ActivityThread.java:7664)
04-18 17:22:30.384  2597  2597 W Bundle  :      at java.lang.reflect.Method.invoke(Native Method)
04-18 17:22:30.384  2597  2597 W Bundle  :      at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
04-18 17:22:30.384  2597  2597 W Bundle  :      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)
```

dennisguse, 2022-04-19:
If I see it correctly, you used use the share track button for a recorded track.
Interesting. Not sure what the actual problem is...

dennisguse, 2022-04-19:
Interesting:
* can you check if you share multiple tracks (from the track list)?
* for individual tracks: does this always crash?

Current hypothesis: there is some issue in the underlying Android method of your device (likely the unmarshaling).
Argument for thIs: we are passing an array (as we might be sharing 1 or more tracks) and if this unmarshals to a string (by Android), the crash is expected.

Code that triggers this problem is in `ShareUtils.newShareFileIntent()` while the crash is by `de.dennisguse.opentracks.TrackRecordedActivity.onOptionsItemSelected(TrackRecordedActivity.java:174)` (which is a subsequent call).

brianjmurrell, 2022-04-19:
Yes, I was trying to share with KDE Connect.  But in inspecting the logs a bit more closely, this looks like it might have been a KDE Connect failure.  I filed [a ticket](https://bugs.kde.org/show_bug.cgi?id=452764) with them on it.

I'll close this here for now.

closed 2022-04-19 by brianjmurrell