OpenTracksApp/OpenTracks #1195: add more granular distance intervals

frasty, 2022-04-20:
Currently OpenTracks only allows to select a minimum distance interval of 1km. This is way too much for activities like swimming. Would be nice if the user could select a more appropriate distance interval of 100m or so, at least for swimming and open water swimming.

dennisguse, 2022-04-21:
What do you mean?
We have 1m until 100m with a default of 10m.

Or do we have some wrong translation?
For which language?

[image]

pstorch, 2022-04-21:
Maybe the distance interval of voice announcements?

frasty, 2022-04-21:
Yes, I mean distance interval of voice announcements and split intervals in the stats. Both have a minimum selectable distance of 1km which is useless for workouts in the range of 2-3km, a fair common distance for swimming workouts, I believe.

dennisguse, 2022-04-21:
I thought that would be quick and easy, but it affects the translations `value_integer_kilometer` and `value_integer_mile` that do not yet support decimal digits...

closed 2022-04-23 by dennisguse

serpico, 2022-04-24:
#1159

dennisguse, 2022-04-24:
@serpico ?

serpico, 2022-04-24:
I haven't seen the look/screenshot of the last PR but I was thinking a slider button might cover a wider range of options, from 100 m to, maybe, 5 km...how many options would be too many ? ( 100, 200, 500m, 1km, 5km )

dennisguse, 2022-04-24:
We can add as many as we want.
Currently, it is a list and you will just need to scroll longer ;)

serpico, 2022-04-24:
Thanks 

> @serpico ?

I mentioned the issue because it sounded about the same parameters : How often do we hear the voice announcement ( every X meters/kilometers and/or every Y minutes ) and the _split intervals in the stats_ ( if announced every kilometer then averaged over the last kilometer ; if announced every minute then averaged over the last minute, if announced every 2 minutes then averaged over the last 2 minutes, and so on...)

commit 7458ea6b (2022-04-23): Voice announcements and Interval: add sub km/miles options.

Fixes #1195.
commit 3815ee6c (2022-04-23): Voice announcements and Interval: add sub km/miles options.

Fixes #1195.