OpenTracksApp/OpenTracks #1204: Low maximal speed

vsevjednom-cz, 2022-04-24:
**Describe the bug**
Start tracking and ride your bike down the hill. Maximal speed calculation is weird. I'm checking GPX in PHP and I'm using this code:
```
		$thistime = new DateTime($point->time);
		$lasttime = new DateTime($lastpoint->time);
		$movetime = $thistime->format("U.u")-$lasttime->format("U.u");
		$speed = round((($distance/1000) / $movetime)*3600, 1);
```
Example: My result of calculation in PHP is 60 km/h but OpenTracks shows only 45 km/h. It's not only chance, I see it every day. I'm also using `ol-ext` for `OpenLayers`. It's javascript and there's speed difference too against OpenLayers (about 15 km/h). There's something wrong with calculation in OpenTracks.

**To Reproduce**
1. Start tracking.
2. Ride your bike down the hill.
3. See the maximal speed in OpenTracks.
4. Export GPX and check the speed in PHP/javascript/something.

**Technical information**
 - Device: Honor 8S KSA-LX9
 - OS: Android 9
 - OpenTracks version: v3.26.1

dennisguse, 2022-04-24:
You can try if #1185 solves this issue for you.
Until now we use the GPS provided speed estimate, which is very noisy.
#1185 changes this to actually use the traveled distance (which we already used for moving avg speed).
You can just import a track and see what the result is ;)

Duplicate of #1003.

closed 2022-04-24 by dennisguse

vsevjednom-cz, 2022-06-10:
> You can try if #1185 solves this issue for you.

I don't know because:
```
error: cannot find symbol
            ChartPoint point = new ChartPoint(trackStatistics, trackPoint, trackPoint.getSpeed(), smoothedAltitude, chartByDistance, viewBinding.chartView.getMetricUnits());
                                                                                                                                                          ^
  symbol:   method getMetricUnits()
  location: variable chartView of type ChartView
```

dennisguse, 2022-06-12:
We will remove the speedbuffer and then continue from there.