OpenTracksApp/OpenTracks #1210: Android 13 adapations

dennisguse, 2022-05-03:
New things in Android 13: https://developer.android.com/about/versions/13/features

https://developer.android.com/about/versions/13/summary

- [x] Themed icon
- [ ] Upgrade targetSdk and compile to 33 (to be released)
- [ ] Check if we need POST_NOTITIFICATIONS
- [ ] Check we need block non-matching intent filters
- [ ] Check if we can use developer downgradable permissions
- [ ] Check if we can use new photo picker

dennisguse, 2022-07-01:
Regarding POST_NOTIFICATION: without this permission, our services create a notification, but it won't be shown anywhere.
We now have two options (on Android13): 
a) that's okay and users will have less clutter
b) we think this notification is important (to show that a recording is running; users can use it to go back to OpenTracks)
Then we have to request this permission from the user (and s/he has to approve - I guess)

@rgmf @pstorch Any opinions?
PS/ will affected almost nobody until mid-2024 :sunglasses: 
(I am still on Android 11 ><)

pstorch, 2022-07-02:
If it is annoying to ask for permission, we can make it optional (again another setting). On the other hand the user might be asked only once.

dennisguse, 2022-07-02:
I am thinking about building some welcome screen/ carousel where we can give some basic intro and get all the permissions.
Currently, we ask on first startup already for Bluetooth and Location permission; adding a third doesn't improve it.

rgmf, 2022-07-02:
> I am thinking about building some welcome screen/ carousel where we can give some basic intro and get all the permissions. Currently, we ask on first startup already for Bluetooth and Location permission; adding a third doesn't improve it.

That sounds like a good idea as I think is important the notification and permissions are getting more complex and uncomfortable.

closed 2022-08-17 by dennisguse