OpenTracksApp/OpenTracks #1215: Please enable StopRecording intent

sgofferj, 2022-05-06:
**Is your feature request related to a problem? Please describe.**
I'm trying to stop recording via Tasker. StartRecording works, StopRecording does not. It's commented out in the manifest file at least.

**Describe the solution you'd like**
I'd like the StopRecording intent to work.

**Describe alternatives you've considered**
Manually stopping the recording but that's defeats the purpose of automating with Tasker.
**Additional context**
None

dennisguse, 2022-05-08:
@sgofferj If you have time, you can test #1216 - I need some feedback here ;)
Additional advantage it comes with the new UI :sunglasses:

sgofferj, 2022-05-08:
@dennisguse Sure, where would I find the apk?

dennisguse, 2022-05-08:
It is build by Github Actions: https://github.com/OpenTracksApp/OpenTracks/runs/6340123194?check_suite_focus=true
You will find the link also in the PR if you look for nightly build.

dennisguse, 2022-05-08:
@sgofferj feature will be in tomorrows nightly: https://fdroid.storchp.de/fdroid/repo/

sgofferj, 2022-05-09:
Works for me on Android 8.1 with the latest Tasker.
May I suggest a clarification for the docs? For people who are not deeply familiar with the inner workings of Android, it might be helpful to clarify that only the package name changes with the build, not the class name. I first put .nightly also in the class name and it didn't work.

dennisguse, 2022-05-09:
@sgofferj good idea!
Just edit the readme as you seem fit.

dennisguse, 2022-05-11:
@sgofferj might even be broken completely...
Does it work if stop in the app?