OpenTracksApp/OpenTracks #1219: 3.28.0: auto-export doesn't work

vsevjednom-cz, 2022-05-09:
**Describe the bug**
Auto-export doesn't work after stop tracking. And app crashes sometimes immediately, sometimes after tens of seconds.

**To Reproduce**
1. Enable auto-export GPX in settings.
2. Start tracking.
3. Stop tracking.
4. There is not exported GPX and app crashes sometimes.
5. There is not any error in log about export.

**Technical information**
 - Device: Honor 8S KSA-LX9
 - OS: Android 9
 - OpenTracks version: v3.26.1

vsevjednom-cz, 2022-05-09:
And... these errors are reason why do this: https://github.com/OpenTracksApp/OpenTracks/issues/824

vsevjednom-cz, 2022-05-09:
And... these errors are reason why to do this: https://github.com/OpenTracksApp/OpenTracks/issues/824

dennisguse, 2022-05-09:
Do you export to a shared folder (like NextCloud)?
Does the same appears for KML?
Is this crash new in v3.28.0 or did it appear before already?

vsevjednom-cz, 2022-05-09:
Uh... There was this error again and again this afternoon but I can't repeat it right now. Everything works without problem.

Export is only into the directory in my phone. It's new crash in 3.28.0 but it's gone 🤪. I don't understand it but I'm closing this issue and I'll open it in future maybe.

dennisguse, 2022-05-11:
Seems to be broken...

https://github.com/OpenTracksApp/OpenTracks/issues/1215#issuecomment-1124007123

dennisguse, 2022-05-11:
Just checked: works for me :)

closed 2022-05-11 by dennisguse

vsevjednom-cz, 2022-05-11:
> Seems to be broken...

The problem seems to be that OpenTracks is not completely shut down and started after the update.