OpenTracksApp/OpenTracks #1224: AndroidRuntime: java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for android.content.AttributionSource@72a9e08: GattService registerClient

brianjmurrell, 2022-05-10:
Just upgraded to LineageOS 19.1 and with that, OpenTracks was crashing with:

```
05-10 16:33:07.082 23704 23704 D AndroidRuntime: Shutting down VM
05-10 16:33:07.083 23704 23704 E AndroidRuntime: FATAL EXCEPTION: main
05-10 16:33:07.083 23704 23704 E AndroidRuntime: Process: de.dennisguse.opentracks.debug, PID: 23704
05-10 16:33:07.083 23704 23704 E AndroidRuntime: java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for android.content.AttributionSource@72a9e08: GattService registerClient
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.os.Parcel.createExceptionOrNull(Parcel.java:2425)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.os.Parcel.createException(Parcel.java:2409)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.os.Parcel.readException(Parcel.java:2392)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.os.Parcel.readException(Parcel.java:2334)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.bluetooth.IBluetoothGatt$Stub$Proxy.registerClient(IBluetoothGatt.java:2492)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.bluetooth.BluetoothGatt.registerApp(BluetoothGatt.java:875)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.bluetooth.BluetoothGatt.registerApp(BluetoothGatt.java:846)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.bluetooth.BluetoothGatt.connect(BluetoothGatt.java:943)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.bluetooth.BluetoothDevice.connectGatt(BluetoothDevice.java:2763)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.bluetooth.BluetoothDevice.connectGatt(BluetoothDevice.java:2712)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.bluetooth.BluetoothDevice.connectGatt(BluetoothDevice.java:2683)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.bluetooth.BluetoothDevice.connectGatt(BluetoothDevice.java:2657)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.bluetooth.BluetoothDevice.connectGatt(BluetoothDevice.java:2635)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at de.dennisguse.opentracks.sensors.BluetoothConnectionManager.connect(BluetoothConnectionManager.java:131)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at de.dennisguse.opentracks.sensors.BluetoothRemoteSensorManager.connect(BluetoothRemoteSensorManager.java:166)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at de.dennisguse.opentracks.sensors.BluetoothRemoteSensorManager.-$$Nest$mconnect(Unknown Source:0)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at de.dennisguse.opentracks.sensors.BluetoothRemoteSensorManager$1.onSharedPreferenceChanged(BluetoothRemoteSensorManager.java:82)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at de.dennisguse.opentracks.settings.PreferencesUtils.registerOnSharedPreferenceChangeListener(PreferencesUtils.java:79)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at de.dennisguse.opentracks.sensors.BluetoothRemoteSensorManager.start(BluetoothRemoteSensorManager.java:124)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at de.dennisguse.opentracks.services.handlers.TrackPointCreator.start(TrackPointCreator.java:59)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at de.dennisguse.opentracks.services.TrackRecordingService.startSensors(TrackRecordingService.java:214)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at de.dennisguse.opentracks.services.TrackRecordingService.startRecording(TrackRecordingService.java:201)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at de.dennisguse.opentracks.services.TrackRecordingService.startNewTrack(TrackRecordingService.java:174)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at de.dennisguse.opentracks.TrackRecordingActivity.lambda$new$0$de-dennisguse-opentracks-TrackRecordingActivity(TrackRecordingActivity.java:92)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at de.dennisguse.opentracks.TrackRecordingActivity$$ExternalSyntheticLambda7.onConnected(Unknown Source:2)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.setTrackRecordingService(TrackRecordingServiceConnection.java:144)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.onServiceConnected(TrackRecordingServiceConnection.java:159)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.app.LoadedApk$ServiceDispatcher.doConnected(LoadedApk.java:2077)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.app.LoadedApk$ServiceDispatcher$RunConnection.run(LoadedApk.java:2110)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.os.Handler.handleCallback(Handler.java:938)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:99)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.os.Looper.loopOnce(Looper.java:201)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:288)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:7870)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
05-10 16:33:07.083 23704 23704 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)
05-10 16:33:07.086  1028 23778 I DropBoxManagerService: add tag=data_app_crash isTagEnabled=true flags=0x2
05-10 16:33:07.086  1028  1951 W ActivityTaskManager:   Force finishing activity de.dennisguse.opentracks.debug/de.dennisguse.opentracks.TrackRecordingActivity
05-10 16:33:07.087  1028  1951 W ActivityTaskManager:   Force finishing activity de.dennisguse.opentracks.debug/de.dennisguse.opentracks.TrackListActivity
```

I looked at permissions for OpenTracks and "Nearby Devices" was not enabled.  Enabling that stopped the crash.

But crashing on a missing permission doesn't seem very friendly.

dennisguse, 2022-05-10:
We request this permission in the settings, but might not check it for recordings.
So, was this permission automatically removed?
Seems to be one of the Android 12 security features.

brianjmurrell, 2022-05-10:
I don't recall removing it.  It could be new in A12 and might need OpenTracks to prompt to enable it on A12 devices perhaps?

brianjmurrell, 2022-05-11:
I did have a different app that uses Bluetooth that I had already installed on Android 11 ask for a new "nearby devices" permission the first time I started it on Android 12.  That seems like the way to go I guess.

dennisguse, 2022-05-11:
That fits. We need to check presence of the permission when starting a track and request if needed.

dennisguse, 2022-05-11:
Cool aspect: this permission was introduced in Android 12 - so we only have trouble for (a) upgrades or (b) automatically/manually revoked permissions.

dennisguse, 2022-05-11:
And there is another problem: we request the GPS permission in TrackRecordingActivity, but this is now too late.
The recording is actually triggered now from TrackListActivity due to the implementation of #1103.

commit 449fc4f9 (2022-05-13): Refactor permission requests.

Fixes #1224.
commit e409b315 (2022-05-19): Refactor permission requests.

Fixes #1224.
closed 2022-05-19 by dennisguse