OpenTracksApp/OpenTracks #1232: v3.28.0: NPE when generating filenames (export/ share)

dennisguse, 2022-05-13:
Template for TrackFilenameGenerator might be null.

```
java.lang.RuntimeException: 
  at android.os.AsyncTask$4.done (AsyncTask.java:415)
  at java.util.concurrent.FutureTask.finishCompletion (FutureTask.java:383)
  at java.util.concurrent.FutureTask.setException (FutureTask.java:252)
  at java.util.concurrent.FutureTask.run (FutureTask.java:271)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
  at java.lang.Thread.run (Thread.java:923)
Caused by: java.lang.NullPointerException: 
  at java.lang.StringBuilder.<init> (StringBuilder.java:113)
  at de.dennisguse.opentracks.io.file.TrackFilenameGenerator.format (TrackFilenameGenerator.java:49)
  at de.dennisguse.opentracks.io.file.TrackFilenameGenerator.format (TrackFilenameGenerator.java:45)
  at de.dennisguse.opentracks.io.file.TrackFilenameGenerator.getExample (TrackFilenameGenerator.java:97)
  at de.dennisguse.opentracks.io.file.TrackFilenameGenerator.isValid (TrackFilenameGenerator.java:84)
  at de.dennisguse.opentracks.settings.PreferencesUtils.getTrackFileformatGenerator (PreferencesUtils.java:580)
  at de.dennisguse.opentracks.util.ExportUtils.getExportDocumentFileUri (ExportUtils.java:111)
  at de.dennisguse.opentracks.util.ExportUtils.exportTrack (ExportUtils.java:63)
  at de.dennisguse.opentracks.io.file.exporter.ExportService.onHandleWork (ExportService.java:62)
  at androidx.core.app.JobIntentService$CommandProcessor.doInBackground (JobIntentService.java:396)
  at androidx.core.app.JobIntentService$CommandProcessor.doInBackground (JobIntentService.java:387)
  at android.os.AsyncTask$3.call (AsyncTask.java:394)
  at java.util.concurrent.FutureTask.run (FutureTask.java:266)
```

dennisguse, 2022-05-13:
@rgmf @pstorch Sadly, that is an ugly bug.
I am not sure if people can set the template via the UI: best guess it crashes in the preferences.
But I can not test it as I have no old installation with data and the template not set.

What about creating a new release on Sunday (will include new UI)?
And what is the new version number? 3.29.0 or 4.0.0?

closed 2022-05-13 by dennisguse

pstorch, 2022-05-13:
Very strange. Can't imaging how it could be null.

dennisguse, 2022-05-13:
Me being stupid 🤣

We have PreferencesUtils.getString() and resources.getString(). They are not the same....
Guess it was too late.

pstorch, 2022-05-13:
Ah, now I get it. You tried to read the DEFAULT from the preferences instead of the resources. Very tricky.

dennisguse, 2022-05-13:
Did the same for the units.
At least I was consistent ;)

PS got it via PlayStore. Affected at least 5 people already.

pstorch, 2022-05-13:
> PS got it via PlayStore. Affected at least 5 people already.

That is the good part of the PlayStore, to get the Exceptions. I recently thought about how to get them outside the PlayStore without some extra lib or 3rd party service. Then I saw how the Nextcloud app is doing it. If it crashes it handles the Exceptions itself, opens a special Activity where the user can see it and decide to send the crash report to a GitHub issue directly. 

It registers an `Thread.UncaughtExceptionHandler` here:
https://github.com/nextcloud/android/blob/f56b926f259425a7765cf56f8f15a56bfac7d897/app/src/main/java/com/owncloud/android/MainApp.java#L242

dennisguse, 2022-05-13:
Good idea!
How would we get the info? Email?

The major advantage of PlayStore is that they aggregate it.
There are something like 180 users (actually relatively stable).

pstorch, 2022-05-14:
What Nextcloud is doing is that they start an Intent with the GitHub new issue URL and add the crash report to it 
Unfortunately no aggregation. I wanted to report a crash, landed on the new issue page and submitted it. Only to later find out, that this crash was already reported, so I closed it as duplicate.
Showing only the Stack trace with the possibility to copy it, would be a first step. How often did we ask for a logcat?

dennisguse, 2022-05-14:
@pstorch Sounds great!

pstorch, 2022-05-15:
The new nightly has this feature. Can be tested via Menu Advanced / Test crash report.
I like it.

dennisguse, 2022-05-15:
Looks great! PR welcome ;)

commit e4211642 (2022-05-13): Bugfix: fileformat template was not loaded (NPE).

Fixes #1232.