OpenTracksApp/OpenTracks #1233: TrackRecordingService: remove pause functionality

dennisguse, 2022-05-13:
We don't need the pause functionality anymore, so we can rip out some technical complexity. :partying_face:

dennisguse, 2022-05-13:
Original plan by @rgmf ; and we had a commit for this somewhere.

vsevjednom-cz, 2022-05-16:
😲😲😲
I'm using pause often.

rgmf, 2022-05-16:
> astonishedastonishedastonished I'm using pause often.

Don't worry, you still can "pause" OT. When you need to pause then stop the track and OT will show you an Activity with some stats. From there you can continue with your track or simply stop it.

In short: from user perspective pause functionality still works.

dennisguse, 2022-05-16:
... but it will simplify TrackRecordingService a lot - and simple is good :)

Only difference: the new implementation will not keep the sensors on (like GPS) when not recording.
Advantage: saves energy
Disadvantage: re-connect or acquiring the fix again might take some time.

vsevjednom-cz, 2022-05-16:
> the new implementation will not keep the sensors on (like GPS) when not recording.

Yesss 🎉😋.

vsevjednom-cz, 2022-05-23:
Pause button is away. Select unpause from menu is too complicated. But it can be like this.

There's short time of wait for pause/stop. I'm affraid of stop tracking by mistake. And app crashes and export nothing. I'll create new issue about it today.

dennisguse, 2022-05-23:
@vsevjednom-cz I am confused.
You can just continue a recording from the 2nd screen (TrackStoppedActivity) or decide to no to.
Then you can still resume via the overflow menu in TrackRecordedActivity.
What's the problem?

PS/ export was broken in v3.28.0 (never was build by F-Droid as I tagged the wrong commit).

closed 2022-05-23 by dennisguse

vsevjednom-cz, 2022-05-23:
I don't see `TrackStoppedActivity` because crash 🤪.

commit 180d29b3 (2022-05-13): TrackRecordingService: remove pause functionality.

Fixes #1233.
commit 71a25a86 (2022-05-22): TrackRecordingService: remove pause functionality.

Fixes #1233.
commit 5ec8d056 (2022-05-22): TrackRecordingService: remove pause functionality.

Fixes #1233.
commit 9a9be2ad (2022-05-23): TrackRecordingService: remove pause functionality.

Fixes #1233.