OpenTracksApp/OpenTracks #1235: When recording TrackListActivity should open TrackRecordingActivity

rgmf, 2022-05-14:
**Describe the bug**
When I was testing #1234 I stumbled upon these two bugs. See how to reproduce to understand the bug.

PS1/ It's not related to #1234 
PS2/ It was introduced between #1155 and last commit.

**To Reproduce but 1**
1. Start a new recording.
2. OT is in TrackRecordingActivity. Press back button.
3. OT is in TrackListActivity. Click on current recording track.
4. OT opens TrackRecordedActivity instead of TrackRecordingActivity.

**To Reproduce but 2**
1. Start a new recording.
2. Back to TrackListActivity.
3. Stop recording from TrackListActivity.
4. The recording is not finished.

dennisguse, 2022-05-14:
1: I introduced it in 9d64989ef1d64b7d9146f3643ab8ceb39a2e9c58 in TrackRecordedActivity
We have two different newIntents and I "unified" them into one.
And that does not work. 
`
2: don't know

dennisguse, 2022-05-15:
TrackListActivity is now opening the correct activity, but I don't understand why this fixes 2 as well :shrug:

commit 16a29c5e (2022-05-15): Bugfix: for a recording Track TrackListActivity needs to open TrackRecordingActivity.

Introduced in 9d64989ef1d64b7d9146f3643ab8ceb39a2e9c58

Fixes #1235.
closed 2022-05-15 by dennisguse