OpenTracksApp/OpenTracks #1236: Cannot parse GPX files with null time

SamWhited, 2022-05-15:
**Describe the bug**

When opening a GPX file with the time set to "null", the import fails.

**To Reproduce**

Open the attached file. You will see "Completed with one error" and the track will not exist on your home page. If you hit show error you will see "Unable to prase time: null".

**Technical information**
 - Device: Asus ZenPhone 8
 - OS: Lineage OS Android 12 Nightly
 - OpenTracks version: v3.27.0 F-Droid

The following file demonstrates the problem (gzipped so that GitHub will allow the upload). As you ca see, no point has a timestamp.

[Sope_Creek_modified.gpx.gz](https://github.com/OpenTracksApp/OpenTracks/files/8694902/Sope_Creek_modified.gpx.gz)

I have worked around the problem temporarily with the following patch, but its behavior was just a workaround and not ideal:

```diff
diff --git a/src/main/java/de/dennisguse/opentracks/io/file/importer/GpxTrackImporter.java b/src/main/java/de/dennisguse/opentracks/io/file/importer/GpxTrackImporter.java
index 22dc421995d8..1840e3296ce8 100644
--- a/src/main/java/de/dennisguse/opentracks/io/file/importer/GpxTrackImporter.java
+++ b/src/main/java/de/dennisguse/opentracks/io/file/importer/GpxTrackImporter.java
@@ -280,7 +280,8 @@ public class GpxTrackImporter extends DefaultHandler implements XMLImporter.Trac
                 zoneOffset = parsedTime.getOffset();
             }
         } catch (Exception e) {
-            throw new ParsingException(createErrorMessage(String.format(Locale.US, "Unable to parse time: %s", time)), e);
+            parsedTime = OffsetDateTime.now();
+           // throw new ParsingException(createErrorMessage(String.format(Locale.US, "Unable to parse time: %s", time)), e);
         }
 
         TrackPoint trackPoint = new TrackPoint(TrackPoint.Type.TRACKPOINT, parsedTime.toInstant());
```

dennisguse, 2022-05-15:
In OpenTracks each TrackPoint requires a time - it is used to compute the statistics.

What is your use case?

SamWhited, 2022-05-15:
A few things:

I have OpenTracks set to the default app to open GPX files on my device. If it can't open them initially, I can't use its built in app picker to open them in something that can display them. If I set something else as the default app, I can't open exported tracks that do have time information easily in OpenTracks, so having it at least be able to import and display valid files without stats is useful.

I also have a collection of tracks, eg. for group bike rides, that I store as part of my OpenTracks library. Some of them I recorded, but others I downloaded from various websites. Having these at the bottom lets me start recording, then quickly open the route in my mapping app all from one interface. Some of them will have time info (the ones recorded in OpenTracks), some may or may not (the ones downloaded from the internet).

Another reason I like being able to open tracks without time is that *some* stats like length are still available. OpenTracks is the only app that I have which will just tell me the length of a track, and that's very useful before going on a long ride.

Finally, even if we don't want to allow importing tracks with no times, this is still a UX issue. If I'm not a software person and I download a track from the internet and it opens with OpenTracks, I'm going to be surprised when I get an obscure error about some programmy-null-thing I don't understand when the track opened just fine on whatever thing I downloaded it with. Instead, it would be nice if it could say "this track has no stats and isn't supported by OpenTracks" or something that explains to the user why the import of a valid file that works in their other apps fails (however, I really hope you'll consider letting me add them to my library either way and just have it show the stats as unavailable or something).

Thanks for reading and responding to my bug report, and for this great app, I love going on a bike ride with OpenTracks!

dennisguse, 2022-05-15:
Sounds reasonable.

If you have time to this, go ahead.
Sadly it won't be that simple as the time is used at several places (like diagrams, exports, statistics etc).
Getting all this right is non-trivial, multi-hour exercise.

dennisguse, 2025-01-07:
Luckily, this functionality  actually just becomes available in OSMDashboard: https://github.com/OpenTracksApp/OSMDashboard/issues/392

@pstorch FYI.

closed 2025-01-07 by dennisguse