OpenTracksApp/OpenTracks #1238: PineTime smartwatch support

aha999, 2022-05-19:
**Is your feature request related to a problem? Please describe.**
I couldn't find a way to connect and open source smartwatch [PineTime](https://www.pine64.org/pinetime/) the Opentracks, so the heart rate monitoring would be visible.

**Describe the solution you'd like**
If possible, it would be great if PineTime could be used as a heart BPM monitor with OpenTracks.

dennisguse, 2022-05-19:
This already works.
PineTime doesn't send that it supports heartrate and it's advertisement packets, so the filter (settings -> bluetooth sensors -> only show Bluetooth devices that ann...) must be disabled before searching for the device.
Anyhow, the data is only reported by the PineTime if it's screen is on.
And I found it to be quite imprecise.

Duplicate of #1063 (also more infos).

closed 2022-05-19 by dennisguse

aha999, 2022-05-19:
Oh, sorry, i didn't find that duplicate because it is spelled together: PineTime, so please edit that title.

dennisguse, 2022-05-19:
Done; no worries.