OpenTracksApp/OpenTracks #1256: Provide Github downloadable release

SB13X, 2022-05-28:
I often prefer getting my apps directly from devs and for the majority of my apps I use a feed app to track releases using githubs ability to append .atom to the release URL to notify of new releases. It would be great if you could provide a build on your release page too for this purpose to remove the dependency on F-Droid or using your repo page without the ability to track updates.

dennisguse, 2022-05-28:
What about the nightly builds?

SB13X, 2022-05-28:
Would it not be possible to make use of the unattended app update API the same way Signal and Newpipe for example manage theirs?

dennisguse, 2022-05-28:
How do they do it?

OpenTracks gives you four options (beside self compiling):
* GitHub actions: if you are logged in an APK can be downloaded for each commit
* F-Droid nightly (hosted by @pstorch)
* F-Droid (official)
* PlayStore

dennisguse, 2022-05-28:
PlayStore releases are build by me.

Quick question: why should the builds by the original devs be better? I would not trust my machine too much myself - so why would you trust it?

Problem with F-Droid: it is kind of slow (5-10 days delay is normal), but that's not a real problem.

And I would not use the GitHub Actions builds for anything critical (as it is not my Infrastructure that builds it).

SB13X, 2022-05-29:
https://nitter.net/DanielMicay/status/1530722999040675840

dennisguse, 2022-05-29:
Same old argument: it boils down to who's infrastructure you want/ have to trust.
If you don't like F-Droid, then don't use it.
You can also host a separate F-Droid repo and build all the applications yourself - the scripts are there and it is not :rocket: science.

If you want build from the developers of OpenTracks:
* you may use the provided nightly (separate F-Droid repo maintained by @pstorch); allows manual download as well
* or you may use the PlayStore build (provided by me).
  
Or you use the one that is build by Github for almost every commit: https://github.com/OpenTracksApp/OpenTracks/actions?query=workflow%3A%22Build+nightly%22++

But as long as we don't have reproducible builds (#418), you have to trust somebody.
In your case, I would rather recommend to build OpenTracks yourself.
Then you have to trust yourself :sunglasses:

closed 2022-05-29 by dennisguse

basings, 2024-01-01:
You can't use e.g. obtainium or other apps to fetch the nightly builds from github. Usually apps are published under /releases/ but not the nightlies

dennisguse, 2024-01-01:
@basings our GitHub nightly build was meant for testing only. 
It is just a convenience thing. 

You can use @pstorch repo, which contains build for the main branch. 

https://fdroid.storchp.de/fdroid/repo?fingerprint=99985A7E73DCB0B16C9BDDCE7A0B4996F88068AE7C771ED53E217E69CD1FF196