OpenTracksApp/OpenTracks #1269: Feedback UI: Hint to scroll

Der-Flip, 2022-05-31:
based on #1264 

**Is your feature request related to a problem? Please describe.**
In order to signal to the user the list is scrollable it is common (and good!) practice to not allow a whole number of elements to fit onto the screen at any point thus cutting off 2/3 - 1/2 of the last or first item respectively when scrolled to the very top or very bottom.
Currently (at least on my device [Redmi9C, screen height: 1600], others may vary depending on screen size of course, in that case there isn't really that much on could do)

**Describe the solution you'd like**
The easiest way to achieve that is to increase the height of the menu bar. this would also allow for slightly larger icons of the menu items within the menu bar as described in #1268 
The size of the record button may need to be adjusted ever so slightly according to menu-bar height and menu icon size.

rgmf, 2022-06-01:
It seems like a coincidence that the last item is cut off at any point, doesn't it? It depends on the size of the screen and other characteristics we can't control.

For instance, in my phone (Pixel 4a), the last item is cut off.

I don't know if I'm overlooking something :thinking:

Der-Flip, 2022-06-01:
yeah, this is what I suspected to be honest. Oh well, it is what it is.
(And to be fair, I can't really remember what it was like with the old UI, anymore.)

closed 2023-07-26 by dennisguse