OpenTracksApp/OpenTracks #1275: Missing important voice announcement settings

delcier, 2022-06-03:
@spacecowboy once posted this image (https://github.com/OpenTracksApp/OpenTracks/pull/1132) that includes additional settings for voice announcements:

[image]

As far as I can tell, these should have made their way into the app in March. However, I don't see these options on my end, only "Report heart rate". By default, the announcement only informs me about the distance. Unfortunately, this info is pretty useless in my case (I need average pace) and I'm sure other people could benefit from additional customization as well.

Is this a bug, or are/aren't there plans to implement these additional settings?

dennisguse, 2022-06-04:
@delcier You are right.

It could be that @spacecowboy never opened a PR for this.

It seems as if this branch contains the changes: https://github.com/spacecowboy/OpenTracks/tree/configurable_announce
This commit adds toggles: https://github.com/OpenTracksApp/OpenTracks/commit/2f29da2046cb12f3492ef8e1fa8ff4a63ee38372

dennisguse, 2022-06-06:
I included the changes; only some minor adjustments were necessary.
Changes will be in the next nightly build.

closed 2022-06-06 by dennisguse

delcier, 2022-06-13:
Thanks, guys!