OpenTracksApp/OpenTracks #1278: Elevation gain way off

shoelacesrcool, 2022-06-09:
**Describe the bug**
Lately when I've used the app, the elevation gain on the main screen has been way off. elevation on profile screens looks fairly close.

I tracked using AllTrails at the same time to rule out my GPS and it tracked gain appropriately.

Example, on a 2.42 mile hike today, it shows elevation gain of 18.6 miles. Actual elevation gain should be close to 300-400 feet.

![Screenshot_20220608-170415](https://user-images.githubuserconte
[image]
nt.com/62978747/172737632-ae37a307-6f82-4ca9-97d0-b7974f59185e.png)

If applicable:
* add screenshots
* add logcat output

**Technical information**
 - Device: [Pixel 5a]
 - OS: [android 12]
 - OpenTracks version: [4.0.0]

dennisguse, 2022-06-09:
Can you check in the intervals tab, when too much was measured?
If it is in the very beginning: fixing #1254 might be a solution.

shoelacesrcool, 2022-06-09:
It shows 10.2 miles in the first mile, 6.7 in the second, and 1.7 miles in
the last parial mile.

dennisguse, 2022-06-10:
That sounds like an issue with the barometric sensor in your phone.
#1254 is about the Android OS providing data too early (sensor not completely initialized yet).

What you can try out: 
1. start a recording and just stay idle: gain/loss should stay at zero
2. go for small walk with the phone in your hand (e.g., not in the backpack)
3. repeat 1+2 while the phone is not in a case

PS/ OpenTracks counts elevation gain/loss changes if they are larger than 3m to the previous stored height.
But we only take the data from the barometric sensor; GPS provided elevation is not used.

shoelacesrcool, 2022-06-10:
I determine if my phone stays completely stationary, the gain/loss does not
change. But moving it less than a foot vertically or horizontally will
usually add 10 feet or so of gain/loss. Even squeezing the phone without
moving it gives a change.

It would indeed seem to be an issue with my phone. Sorry for the trouble.

dennisguse, 2022-06-11:
No trouble at all :)

closed 2022-06-11 by dennisguse