OpenTracksApp/OpenTracks #1282: Public API: Additional parameters for StartRecording

sgofferj, 2022-06-13:
**Is your feature request related to a problem? Please describe.**
Not a problem - more a comfort thing. I'm heavily using tasker to automate things. E.g. I have widgets for starting and stopping a dogwalk which will start OpenTracks, tell homeassistant I'm out with the dogs and set the ringing and media volume of the phone to max. When using the StartRecording and StopRecording intents, there is no possibility to set e.g. the workout type.

**Describe the solution you'd like**
I would like to suggest additional parameters for the StartRecording intent e.g. to set the workout type. This way, one can have e.g. different tasker homescreen widgets or generally, other apps can do thing with OpenTracks with a little more granular control.
E.g. starting and stopping a biking workout when unlocking/locking the bluetooth bike lock.

**Describe alternatives you've considered**
The only alternative as of now is to go into OpenTracks and stop an API-started recording from there to get to the save screen and enter parameters.

**Additional context**
n/a

dennisguse, 2022-06-14:
Definitely possible; relevant code: `StartRecording` and `ContentProviderUtils.updateTrack(Track track)`.

dennisguse, 2022-07-16:
@vanous fyi and 
https://codeberg.org/Freeyourgadget/Gadgetbridge/wiki/OpenTracks-API needs an update ;)

vanous, 2022-07-17:
This is cool, thank you, wiki updated. cc: @arjan-s

arjan-s, 2022-07-17:
Hey, that's cool! Thanks @dennisguse !

arjan-s, 2022-07-17:
@dennisguse  I've added support for setting the category of a track in Gadgetbridge, and it seems to work. The category is set correctly, but the track icon still shows the generic icon. Could it be that on non-English phones, OpenTracks expects the translated category name, instead of the English name? If so, that would make correct implementation in Gadgetbridge and other apps pretty difficult. Would it be possible to allow setting the English name, or maybe even a numeric category ID, and use that to show the correct icon on the track?

dennisguse, 2022-07-17:
@arjan-s don't get me started on the icons ;)
Yes, you are right, it we expected localized input there. And category is always localized (even storage + import).

As the category can be anything (aka user defined), it is kind of tricky.
But we should do something about it.
Suggestions?

arjan-s, 2022-07-17:
Sorry... :)
I would suggest to show the icons on the known English categories as well, in addition to the translated categories. The shown category will then still be English instead of translated, which I think is fine, but the correct icon will be shown.

vanous, 2022-07-18:
It seems to me that categories have icons assigned, so there must be some category or icon id. So i would add a Category ID or Icon ID as a possible Intent parameter. This would make it work universally no matter the localization of the device for us. If the Intent provided both Category string and Icon ID, OT would need to decide which one takes precedence, which is fine.
What do you think?

dennisguse, 2022-07-18:
Code is in `TrackIconUtils`, but there is not category id - it is just the list of to be translated string:
```
    private static final int[] AIRPLANE_LIST = new int[]{R.string.activity_type_airplane, R.string.activity_type_commercial_airplane, R.string.activity_type_rc_airplane};
```
Icon id might be this:
```
    private static final String AIRPLANE = "AIRPLANE";
```

PS/ naming could be improved; we use category and actvityType in the code interchangeably.

commit 7bd88742 (2022-07-02): Public API: allow setting track properties (name, category, description) on StartRecording.

Fixes #1282.
commit e1d3d528 (2022-07-03): Public API: allow setting track properties (name, category, description) on StartRecording.

Fixes #1282.
PR #1313 (2022-07-03): Public API: allow setting track properties (name, category, description) on StartRecording)
@sgofferj does this work for you?

Fixes #1282.
commit c4e20239 (2022-07-03): Public API: allow setting track properties (name, category, description) on StartRecording.

Fixes #1282.
closed 2022-08-28 by dennisguse