OpenTracksApp/OpenTracks #1288: Split interval dropdown options are only in metric

andmagdo, 2022-06-17:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

[image]

**Technical information**
 - Device: Google Pixel 6
 - OS: GrapheneOS latest stable version (build SQ3A.220605.009.B1.2022061600)
 - OpenTracks version: v4.0.1 (latest on F-droid)

dennisguse, 2022-06-18:
Issue should be in `IntervalsFragment.setIntervalsDropdownText()`.

rgmf, 2022-06-18:
This is a bug from the begining... I overlooked it.

Intervals dropdown values are always showed as metrics values.

I get it.

rgmf, 2022-06-18:
@andmagdo thank you so much for reporting this bug ;)

commit 895a5c2b (2022-06-18): Bugfixed: DistanceFormatter was using the wrong unitSystem. Fixes #1288.
closed 2022-06-18 by rgmf