OpenTracksApp/OpenTracks #1297: Make the recording of altitude data optional

Mugen36, 2022-06-25:
**Is your feature request related to a problem? Please describe.**
The elevation data on my tracks is completely non-sense. I am running 5 times around the block. And the graph shows that I climbed down 20 meters. And frankly, if one use Opentracks for evening runs, tracking the altitude is not that important anyways.

**Describe the solution you'd like**
Make recording elevation data optional. Implement an option that disables the recording of altitude data.

dennisguse, 2022-06-25:
The absolute elevation from GPS or gain/loss from the barometric sensor?
And not showing it or not storing it or not exporting it?

Where is it shown that you don't want it?

Mugen36, 2022-06-26:
I was referring to the absolute elevation from the gps. I actually do not see any data from the barometric sensor.
The GPS-Beidoo-Glonass location data and the running speed seems fairly accurate. Accuracy usually <5m.
Yet the elevation data is completely garbage. Not only does it show a 20m climb-down where there is none. It is also roughly 40m too high.
I would prefer an option of not storing it, since it is just plainly false.

I see the elevation data in the graphs when I look at the specific tracks.

Mugen36, 2022-06-26:
At the core, I ultimately would blame Google for not putting a better GPS in there Pixels. Yet, since elevation data is usually far less accurate than horizontal location, I can imagine many more people have this issue.

dennisguse, 2022-06-27:
Well no.

1. GPS accuracy for horizontal data is less accurate than vertical data (due to how GPS works)2
2. exported data is in WGS84 as specified by KML/GPX.
   but this is not above ground aka sea level value. That's why OpenTracks converts for the UI the elevation into EGM2008.

And implementing this switch is too much effort; especially maintaining it.
The only thing would be drop data that is not accurate enough as reported by the GPS (horizontal accuracy is provided)

Mugen36, 2022-06-27:
> And not showing it or not storing it or not exporting it?

Well, if not storing and not exporting, are not reasonable requests,
could I ask for an option for not showing elevation data in the graphs?

dennisguse, 2022-06-27:
That would work.
Interested to implement it?

Mugen36, 2022-06-27:
Sure

Mugen36, 2022-06-27:
May I ask, am I the only one who is getting non-sensical altitude values? 

Or am I just the only one who is annoyed by seeing these in the graph?

dennisguse, 2022-06-27:
How would I know :sunglasses: 
We have literally zero monitoring and know only stuff if somebody reports.
Can you share a screenshot of your chart?

Mugen36, 2022-06-27:
Here you go!

4 different days. The track is always the same. After the first half of my track, I turn around and run the same way back home.
I looked up the altitude of my track in Wikipedia. I am always running around 37m to 40m meters above sea level.
[image]
[image]
[image]
[image]

dennisguse, 2022-06-28:
Actually, we just removed the smoothing for the elevation (was only used in the charts) as it was broken #1241.

Mugen36, 2022-06-28:
I dont think smoothing would help much here. The base values are just too far off.
I did tests with different devices and in every case the altitude values were too unreliable to be of any use.
When you use opentracks, do you get any informative altitude data?

Please consider reopening this issue. Implementing it will be fairly easy. And I will definitely look into it although I  am not familiar with the code base and quite pressed for time atm.

dennisguse, 2022-06-29:
@Mugen36 Just re-open it; the issue tracker is a place for everybody :)
The charts were actually configurable in MyTracks - so somewhere in the git history there is a commit for this. I guess I removed while trying to clean up as much as possible

PS/ I don't use the elevation data that much as I live in an area without any mountains.
So, for me it is not an issue that much.
And I don't use the charts for analytics.

Mugen36, 2022-06-29:
> @Mugen36 Just re-open it; the issue tracker is a place for everybody :) 
I dont believe github allows me to reopen issue that were closed by a collaborator.

>

crankc, 2022-07-01:
> 

Sure, i opened a feature request regarding editing summary layouts and hiding unwanted data within.

About the pull request you linked, i'm not sure what to check? I'm not a developer, just an end user, but to my eyes it seems like it might help. I tested my phone with Trail Sense and sure enough, the barometric sensor values change rapidly every time i quickly move the phone, or even squeeze it gently. Now i know why it showed me 45 kilometers of gain after i had it in my pocket while cycling.

dennisguse, 2022-07-01:
@crankc if you signed into Github, you can download the APK for each pull request and install it manually (take a look at the Github Actions and then Nightly).
Not that convenient but it works.

MartinH80, 2023-05-08:
I recently started running again after several years. Happy to have found OpenTracks via fdroid!

My phone is a Pixel 6 and the altitude graphs are between 1 and 8 meters. That could be ok for where I live, but I can't match a the graph exactly with my observed reality.

More important to me is that I'm not interested in altitude and to me having two lines in one graph makes it harder to read exactly that what I'm looking for: my running pace. 

So here another user who would be even more happy with OpenTracks when either (1) in settings the collection or visibility of this altitude data could be configured or (2) in the graphs themselves I could click on e.g. 'height' to hide this series.

I'm not a developer myself, but if I can do anything to help, please let me know!

dennisguse, 2023-05-09:
What about hiding the graph if less than 20m of difference was encountered?

MartinH80, 2023-05-11:
That solution would work for me! (but I have my doubts if this is a proper solution in general, 20m seems arbitrary and might not cover all use cases)

hercek, 2023-12-31:
I vote for an option to hide altitude data in the charts. It would be useful for people who train mostly at the same altitude. For them, it clutters the charts with "garbage" since it is relatively imprecise.

Brandon-Huu, 2024-07-17:
Why is this not an option yet?
I've running on a flat road and open tracks is telling me I climbed up a mountain both ways on my way to and from school

dennisguse, 2024-07-17:
@Brandon-Huu if you have time, you can work on this. Wasn't the highest priority...

Brandon-Huu, 2024-07-17:
o7 I'll see what I can do

grobolom, 2025-01-24:
Toggles for elevation, speed, and heart rate visibility on the charts are now included in the `User Interface` Settings page per #2054.