OpenTracksApp/OpenTracks #1301: Crash when workout triggered through Gadgetbridge

haarp, 2022-06-28:
Hello,

**Describe the bug**
When Gadgetbridge tries to trigger a workout, OpenTracks crashes. When I unlock the phone, I find a crash report waiting for me. Seems like the app is missing a permission? It does not have any denied permissions tho.

`Public API` and `Automatic data transfer` are enabled in the settings.

Gadgetbridge 0.67.1 is connected to a Fossil HR Collider, with `activitiy recognition` set to `ask` for biking. The watch doesn't ask, but that's likely a GB issue.

Both apps were installed via F-Droid.

**To Reproduce**
1. Set up environment as above
2. Work out/bike until the watch recognizes the activity (there will be no indication of it tho)
3. Unlock phone
4. Find crash report

Crash report:

### App information
* ID: de.dennisguse.opentracks
* Version: 4926 v4.0.1

### Device information
* Brand: google
* Device: flame
* Model: Pixel 4
* Id: QQ3A.200805.001
* Product: flame

### Firmware
* SDK: 29
* Release: 10
* Incremental: 23602ad7c2

### Cause of error
```java
Exception in thread "main": java.lang.RuntimeException: Unable to start activity ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.publicapi.StartRecording}: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=de.dennisguse.opentracks/.services.TrackRecordingService }: app is in background uid UidRecord{eea9098 u0a206 TPSL idle change:idle|cached procs:1 seq(0,0,0)}
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3270)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3409)
	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:83)
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2016)
	at android.os.Handler.dispatchMessage(Handler.java:107)
	at android.os.Looper.loop(Looper.java:214)
	at android.app.ActivityThread.main(ActivityThread.java:7356)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:491)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:940)
Caused by: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=de.dennisguse.opentracks/.services.TrackRecordingService }: app is in background uid UidRecord{eea9098 u0a206 TPSL idle change:idle|cached procs:1 seq(0,0,0)}
	at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1616)
	at android.app.ContextImpl.startService(ContextImpl.java:1571)
	at android.content.ContextWrapper.startService(ContextWrapper.java:669)
	at android.content.ContextWrapper.startService(ContextWrapper.java:669)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.startAndBind(TrackRecordingServiceConnection.java:77)
	at de.dennisguse.opentracks.publicapi.AbstractAPIActivity.onCreate(AbstractAPIActivity.java:37)
	at android.app.Activity.performCreate(Activity.java:7824)
	at android.app.Activity.performCreate(Activity.java:7813)
	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1306)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3245)
	... 11 more
```

**Technical information**
 - Device: Pixel 4
 - OS: LineageOS 17.1
 - OpenTracks version: 4.0.2 (F-Droid)

Thanks!

dennisguse, 2022-06-29:
@haarp can you if this solves the issue for you?
https://github.com/OpenTracksApp/OpenTracks/pull/1303

Anyhow, that is not a proper fix for the whole situation.
We need some more tinkering if this works in general.

@pstorch again KUDOS for the stacktrace feature!

haarp, 2022-06-29:
> can you if this solves the issue for you?

Sure. Could you please make an experimental build for me? I don't have the tools for it.

haarp, 2022-06-29:
I can confirm that in [this build](https://github.com/OpenTracksApp/OpenTracks/actions/runs/2582883258), Gadgetbridge can successfully start recording 👍 Thanks!

dennisguse, 2022-06-30:
@haarp Can you check use this build for the next days?
We have to do some changes to accommodate this and it would be good to know if this really works.

vanous, 2022-07-01:
I do not know if it is related, but on my device (Android 11, vendor OnePlus), OpenTracks does not crash (even without this code modification), but fails to get a GPS fix. Once i open OT, fix is acquired right away. What i noticed is that i can "only" grant a Location permission "Allow only while using the app", as opposed to "Allow all the time". Could this be related? (see also https://github.com/OpenTracksApp/OpenTracks/issues/596). When comparing with for example TrailSense, i do have the "Allow all the time" there. Battery optimization is off for OT.

commit b2269b89 (2022-06-30): HACK: this is not a proper fix.

Fixes #1301.
commit 9f6ca83f (2022-08-04): Bug: PublicAPI needs to call startForegroundService (SDK API26+).

Fixes #1301.
closed 2022-08-04 by dennisguse