OpenTracksApp/OpenTracks #1304: Add Bluetooth ANT+ cycling lights support (automatic-on)

ae5960e8-a6fc-491f-b252-898ecf59af95, 2022-06-29:
Bluetooth lights which power-on automatically for bicycles are becoming somewhat common nowadays. Your bike light might have it without you even knowing. Garmin and Bontrager are the top producers of these that I know of. The idea is when you turn your cycle computer on, the lights turn on also, to save hassle turning on 2 or 3 lights manually. [They activate via Bluetooth ANT+](https://support.garmin.com/en-US/?faq=ydVnNiaZskAq0IbuOXmkG6) which OpenTracks supports well already.

Communication with Bluetooth lights would make these features possible:
- Automatically switch on lights when recording starts
- Automatically switch off lights when recording ends
- A software button to cycle through light modes of connected lights (not as important as prior 2 points)

Could this be discussed and considered?

dennisguse, 2022-06-29:
Interesting. If these lights use some vendor-agnostic protocol on top of Bluetooth LE it becomes doable. Implementing vendor-specific protocols is not cool.

That's one aspect. Another is that we then would need some kind of profile Support aka I am road cycling - please use these sensor is and turn on the light etc.

And I would need one of these Bluetooth enabled lamps for testing (my current one are all simple aka one button only).

dennisguse, 2022-11-02:
I believe that is out of scope for now.

closed 2022-11-02 by dennisguse