OpenTracksApp/OpenTracks #1309: Customize track summary layout, hide unneeded data?

crankc, 2022-07-01:
Opening a new feature request on this as suggested in #1297 

Since we already have a way to customize **recording layout** where i can hide gain/loss data for example, would it be possible to customize **track summary layout** as well?

My desire to hide certain data originally came from my Pixel 5's barometric sensor messing up OpenTracks' gain/loss values, but being able to edit the summary layout would be otherwise welcome as well, and it would make sense because there already is partial layout customization support.

dennisguse, 2022-07-01:
Similar to https://github.com/OpenTracksApp/OpenTracks/issues/1297#issue-1284642533

Now I have a problem :)
If the sensor is broken, it doesn't make any sense to collect the data, right?
So, now we have the situation that there is a broken GPS (abs. elevation) and a broken barometer (rel. elevation/ gain).

And for both the solution would be to just not store the data, because it is useless anyhow.
So, technically speaking we would solve #1297 and #1309, if we would just not collect (barometer) or not store (GPS; elevation) the data.

Technically both is it not complicated.
Question is: how to implement the settings properly?
Any suggestions?

crankc, 2022-07-01:
I'm generally in favor of options as long as they don't add irrelevant clutter to an app. In this case i think the toggle(s) would be a good compromise between hiding data that is useless to some and altogether removing a feature that may be useful to others. I do consider recording/showing altitude values a "basic" feature of any outdoor/track recording/activity tracker app, so i wouldn't outright remove it either. Just my take.

dennisguse, 2022-07-01:
My proposal was to not collect/ store the data and make this configurable.

closed 2024-12-29 by dennisguse