OpenTracksApp/OpenTracks #1320: Interval: question to @rgmf

dennisguse, 2022-07-07:
Does the time assignment make sense?

@rgmf 

```
        public Interval(Interval i, double adjustFactor) {
            distance = i.distance.multipliedBy(adjustFactor);
            time = Duration.ofMillis((long) (i.time.toMillis() * adjustFactor));
            time = i.time;
            gain_m = i.gain_m;
            loss_m = i.loss_m;
            avgHeartRate = i.avgHeartRate;
        }
```

rgmf, 2022-07-07:
The second time assignment seems wrong. Let me check it out.

rgmf, 2022-07-09:
Well, it was wrong, the first time assignment was the good one. Anyway, that constructor is never call, so the bug had not impact.

I've removed it.

commit d0b42295 (2022-07-09): Bugfixed: an Interval constructor assigned twice the time with different values. That constructor is not needed, so it's been deleted. Fixes #1320.
PR #1322 (2022-07-09): Cleanup: adjusted Interval uses constructor.
Fixes #1320.

Does this make sense?
commit 04e14960 (2022-07-09): Cleanup: adjusted Interval uses constructor.

Fixes #1320.
closed 2022-07-09 by rgmf