OpenTracksApp/OpenTracks #1329: PlayStore release of 4.0.4 rejected

dennisguse, 2022-07-16:
The 4.0.4 was rejected by the PlayStore due to an impersonation violation - they complain about the icon and the title.
I submit an "Advance Notice Form" - let's see how long it takes.

Until then 4.0.4 will be _not found_ in the PlayStore.
And all future releases may be rejected as well.

PS/ interestingly 4.0.3 is still available, so okay.

[image]

pstorch, 2022-07-16:
This gets more and more rediculous.

rgmf, 2022-07-16:
🤬🤬

pstorch, 2022-07-16:
And they don't tell which app it is accused of impersonating?

pstorch, 2022-07-16:
DirtTracks maybe?
https://play.google.com/store/apps/details?id=com.zihua.android.dirttracks

Or MyTracks from the same dev?

dennisguse, 2022-07-16:
Actually no.
The screenshot contains all the info I got.
I send them feedback using three different channels.

Anyhow, if they are going after copy cats it is not all bad.
Let's see.

pstorch, 2022-07-16:
> Anyhow, if they are going after copy cats it is not all bad.

Yes, that's right. It's just that they don't give enough information and often don't respond to questions.

dennisguse, 2022-07-17:
Jason takes care :)

```
Hi Developer at Dennis Guse,

Thanks for contacting the Google Play team. 

I've received your appeal and I appreciate your patience while I look into it.

I'll let you know as soon as I have any additional information to share. 

Please let me know if you have any questions in the meantime.

Regards,
Jason
The Google Play Team
```

dennisguse, 2022-07-17:
Another name friend: https://play.google.com/store/apps/details?id=app.opentracks&hl=de&gl=US

dennisguse, 2022-07-18:
And we are impersonating MyTracks, but I am not exactly sure which one...
* https://play.google.com/store/apps/details?id=com.zihua.android.mytracks
* https://play.google.com/store/apps/details?id=com.google.android.maps.mytracks
  (de-published Google MyTracks; only visible if you installed it and are logged in into the PlayStore)

[image]

dennisguse, 2022-07-19:
My response (2022-07-18)
```
Hi,

I presume you are referring to Google's MyTracks (unpublished 2016) [1], right?

OpenTracks is a fork of Google's MyTracks and we continue to make and keep it awesome.
The application code is licensed as Apache 2.0 and adhere to this license.
We state this in the relationship in our README.md: https://github.com/OpenTracksApp/OpenTracks#project-history
And also in the application [2] (Settings->About OpenTracks->About OpenTracks).
And somebody even put OpenTracks onto the Wikipedia page of Google's MyTracks [3].

For this reason (and also because Google's MyTracks is unpublished as of 2016), we use the same logo.
Although the color was changed slightly (slightly darker more orange) some releases ago (this was not deployed to the PlayStore).

Since the fork (v2.0.6, commit id 0fad5840f3d752176919ae5f45318706fc6163f3 ), there were 2228 commits done, making OpenTracks what it is.
So, OpenTracks is not really MyTracks anymore as we changed almost everything (beside the logo).
Moreover, the lineage to MyTracks is important, but it it is not making OpenTracks better.

Under these circumstances, we would like to keep the old logo in place.
Would it help to update the newly colored logo for the PlayStore listing?

Best,

Dennis Guse

[1] https://play.google.com/store/apps/details?id=com.google.android.maps.mytracks
[2] Attached to this email; Screenshot_20220718-074022.png
[3] https://en.wikipedia.org/wiki/MyTracks
```

dennisguse, 2022-07-19:
I can at least say, they respond fast - but I am not sure they would fulfill Turing test.
Kind of repetitive... 

Their response (2022-07-18)
```
Hi Dennis Guse,

Thanks for getting back to us.

Unfortunately I'm not able to comment on your planned implementation.

As mentioned previously, your app's icon and title look highly similar to another app that is published on Google Play: My Tracks.

App titles and icons that are so similar to those of existing products or services that users may be misled are not allowed. You could read through the [Impersonation](https://support.google.com/googleplay/android-developer/answer/9888374) policy page for more details and examples of common violations.

Thanks for working with us to fix the policy issue and for your continued support of Google Play.

Regards,
Jason
The Google Play Team
```

dennisguse, 2022-07-19:
What about: `[Your, Our, Their, ' '] Track['', s]`?

To be honest: who cares about the PlayStore release anyhow.
For this, we could just deploy the black (nightly) icon.
Or we could invert the icon (aka make "arrow" white and the background colored).

Changing the name - I would be too lazy to this, just for the PlayStore releases.
As long as no lawyer starts sending letters: WOLO (we only live once).

dennisguse, 2022-07-19:
PS/ if we need a new name, I am all in.
Just not for the Google situation :cloud_with_rain:

rgmf, 2022-07-20:
I cannot understand why name is an issue. Anyway, whatever decision was taken would be good for me.

dennisguse, 2022-08-02:
PlayStore just accepted v4.0.5...
:facepalm: 

Anyhow: fun something learned.

dennisguse, 2022-08-09:
V4.0.6 got rejected again

dennisguse, 2022-08-16:
So, seemingly they also complain about the uploaded the PlayStore metadata aka screenshots, feature graphics, and icon.
I will replace this accordingly with literally anything else ;)

dennisguse, 2022-08-22:
Just replaced the graphics in the PlayStore listing: now also inverted aka white logo on orange background :man_shrugging: 
Let's see if v4.0.6 is now accepted :)

dennisguse, 2022-08-23:
PlayStore just accepted v4.0.6  for publication.
Let's wait for next time.

closed 2022-08-23 by dennisguse