OpenTracksApp/OpenTracks #1332: Pace wrong in plots

marcelmindemann, 2022-07-19:
**Describe the bug**
I think the plotting of my pace against time/distance is broken. I did a run yesterday where my average pacr was around 7 min/km. This is consistent with the breakdown into 1km intervals.

However the graphs show my pace averaging around 8.6 min/km. That's constantly too high when compared with the interval screen.

**Screenshots**
[image]
[image]
[image]

**Technical information**
 - OpenTracks version: 4.0.2

dennisguse, 2022-07-19:
I have no good explanation for this behavior.
Two options:
* the pace computation is broken in the charts
* or the rendering of the chart is broken (code is ancient ;) )

SlidingHorn, 2022-10-04:
I'm not sure if this is what I'm dealing with or not, so before making a new ticket, I'll describe my issue and let you tell me whether it belongs here or in its own issue:

I recently started using OT for my run/walk intervals (had been using it for a while for my cycling), and if I stop for any period of time to rest, adjust my knee brace, etc., it shows up as a *huge* spike in my graph. This results in my Pace plotting having a scale up to 600 min/mi, which makes reading that data on the graph impossible. 

Screenshot: https://i.postimg.cc/nh811MnH/Screenshot-20221003-202857.png

dennisguse, 2022-10-05:
@marcelmindemann just had some time to look I to this. The problem are the labels on the y-axis. It should start with a 0 for pace/speed. The actual plot seems correct when taking this i to account. Labels should be 0, 5, 10, etc.

Don't know exactly why this looks this way for you.

dennisguse, 2022-10-19:
So, this might be relevant: https://github.com/OpenTracksApp/OpenTracks/issues/795
BUT I didn't see a relevant change in the commits (yet).

@marcelmindemann Can you send me an example track to `opentracks-play@dennisguse.de`?
I cannot reproduce the problem locally.
And does this problem only occur with pace or also with speed?

dennisguse, 2023-05-23:
Please re-open if this is still an issue.

closed 2023-05-23 by dennisguse