OpenTracksApp/OpenTracks #1334: No compatible file manager is installed

eachtime, 2022-07-21:
Hi! thank you for this awesome app. I am trying to export opentracks' data from my old phone to import into a new one. Unfortunately I have the "No compatible file manager is installed" message on export. I also tried an adb backup/restore command without success. I would like to try to export and import the data manually but I cant find them on the filesystem. 

**Technical information**
 - Device: MotoG6
 - OS: Android 9
 - OpenTracks version: last version from f-droid

eachtime, 2022-07-21:
I just found a database.db file under /data/data/de.dennisguse.opentracks. I suppose that the GPS data are there, am I right?

eachtime, 2022-07-21:
I tried to convert the tracks table in the database.db file to a csv to import it but no success. I dont really know which format is expected I'll try again later.

dennisguse, 2022-07-21:
So...
CSV import is not supported.

What you can do: use the share feature. 
Go to the track list. 
Select one track (long press).
Go to context menu: select all.
And the share: send it by mail or so.

eachtime, 2022-07-21:
> So... CSV import is not supported.
> 
> What you can do: use the share feature. Go to the track list. Select one track (long press). Go to context menu: select all. And the share: send it by mail or so.

Awesome! It worked perfectly, thank you very much.

closed 2022-07-21 by dennisguse