OpenTracksApp/OpenTracks #1335: exception when view diagram by time or distance

gittehubble128, 2022-07-22:
### App information
* ID: de.dennisguse.opentracks
* Version: 4951 v4.0.3

### Device information
* Brand: HUAWEI
* Device: HWELE
* Model: ELE-L29
* Id: HUAWEIELE-L29
* Product: ELE-L29EEA

### Firmware
* SDK: 29
* Release: 10
* Incremental: 10.1.0.150C431

### Cause of error
```java
Exception in thread "TrackDataHub": java.lang.ArithmeticException
	at j$.time.temporal.Temporal$-CC.m$2(Unknown Source:67)
	at j$.time.Duration.toMillis(SourceFile:1)
	at de.dennisguse.opentracks.chart.ChartPoint.<init>(ChartPoint.java:45)
	at de.dennisguse.opentracks.chart.ChartFragment.onSampledInTrackPoint(ChartFragment.java:193)
	at de.dennisguse.opentracks.data.TrackDataHub.notifyTrackPointsTableUpdate(TrackDataHub.java:368)
	at de.dennisguse.opentracks.data.TrackDataHub.loadDataForListener(TrackDataHub.java:243)
	at de.dennisguse.opentracks.data.TrackDataHub.lambda$registerTrackDataListener$1$de-dennisguse-opentracks-data-TrackDataHub(TrackDataHub.java:187)
	at de.dennisguse.opentracks.data.TrackDataHub$$ExternalSyntheticLambda0.run(Unknown Source:4)
	at android.os.Handler.handleCallback(Handler.java:900)
	at android.os.Handler.dispatchMessage(Handler.java:103)
	at android.os.Looper.loop(Looper.java:219)
	at android.os.HandlerThread.run(HandlerThread.java:67)
```

gittehubble128, 2022-07-22:
viewing Diagram impossible.
Occurs when tab on both diagrams.

dennisguse, 2022-07-22:
How did you create the tracks?
Are they recorded or did you import them?

gittehubble128, 2022-07-22:
It was recorded by the app.

dennisguse, 2022-07-22:
Do you know with which version?
Was it recently?

Did you use some kind of BLE sensor (like heart rate)?

gittehubble128, 2022-07-22:
The Version is included in the app report, it is most recent, but occured
in (my) previous as well.
Today, I recorded 2 tracks, one crashes in diagram, other not.
A particular track recorded with previous Version crashes with current
Version.
No BLE devices attached.

dennisguse, 2022-07-22:
That sounds okayish.
If you can build yourself?
This might fix the problem: https://github.com/OpenTracksApp/OpenTracks/pull/1310

Just export the crashing track and import into the pull request build.

gittehubble128, 2022-07-22:
Unfortunatly I can't build myself.
I just tried uploading the track to gist but gist is not accepting it as
txt (renamed) nor zipped (on my smartphone) for some reason.
So I think I just habe to wait for next Version if it is a known/addressed
bug.
Thank for the fast reaction anyway.

dennisguse, 2022-07-22:
You can send me the track by mail: play-opentracks@dennisguse.de

gittehubble128, 2022-07-23:
I sent two samples, thanks a lot

dennisguse, 2022-07-24:
Exporting and importing (into another OpenTracks installation) solves the issue.
Should be fixed #1310 anyhow.

@gittehubble128 one more thing: KMZ/KML contains more information than GPX.
For backup purposes, I would recommend those instead.

closed 2022-07-24 by dennisguse

gittehubble128, 2022-07-24:
Thanks Dennis, espacially for your awesome qick responses!