OpenTracksApp/OpenTracks #1338: Altitude shown as 'null' despite being recorded

xfarrow, 2022-07-25:
**Describe the bug**
In the tab 'intervals' tab, all the altitude mesurements are shown as null, despite being recorded, because it is shown in the graphs rendered in the following tabs.

**To Reproduce**
1. Start recording
2. Stop recording
3. Go to the 'intervals' tab
4. See error

If applicable:
* add screenshots
* add logcat output

**Technical information**
 - Device: Huawei Mate20 Lite
 - OS: Android 10
 - OpenTracks version: 4.0.3
 - OpenTracks commit id (only for nightly builds): no nightly

dennisguse, 2022-07-25:
Problem is somewhere in StringUtils.getAltitudeParts()

dennisguse, 2022-07-26:
Thanks for the info.
Will be in tomorrows nightly.

commit 78aa5d51 (2022-07-26): Intervals: if gain/loss is not available show "-" instead of null.

Fixes #1338.
closed 2022-07-26 by dennisguse