OpenTracksApp/OpenTracks #1339: Display non-rounded value of altitude on recording screen

edern-appere, 2022-07-26:
Hi there !

I'm sending this issue as a request for enhancement, if you believe it is relevant.

As of now (I'm using 4.0.3) I think there's something wrong with the way altitude is displayed on the recording screen. Once you reach 500m of altitude, it is shown as 0.5 km of altitude. When you are riding (or hiking), it can make a big difference if your altitude is 590m rather than 510m, although OpenTracks displays it the same way. So my proposal would be to stick with non-rounded values ie display 510m if altitude is 510m rather than 0.5 km (provided there's enough space on the screen).

Have a nice day (and nice rides, of course !)

dennisguse, 2022-07-26:
Duplicate of #1319, which should be fixed by PR #1310.
But #1310 uses a beta component, so we where waiting until it got stable (in September?).

@rgmf @pstorch the issue seems to affect people; should we merge #1310 and hope that it works?

edern-appere, 2022-07-27:
Hi Dennis !
Sorry to have made a duplicate bug report. I made a search but the title and description of #1310 did not draw my attention.
No need to hurry to use a beta component instead of a stable one. Make the best for the app in long term.

Have a nice day !

dennisguse, 2022-07-27:
No worries ;)

closed 2022-07-27 by dennisguse