OpenTracksApp/OpenTracks #1342: Apps crashed when starting Activity on Mi Band 5

Haribari, 2022-08-01:
### App information
* ID: de.dennisguse.opentracks
* Version: 4981 v4.0.4

### Device information
* Brand: Gigaset
* Device: GS290
* Model: GS290
* Id: QQ3A.200805.001
* Product: GS290_EEA

### Firmware
* SDK: 29
* Release: 10
* Incremental: eng.root.20220627.160632

### Cause of error
```java
Exception in thread "main": java.lang.RuntimeException: Unable to start activity ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.publicapi.StartRecording}: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=de.dennisguse.opentracks/.services.TrackRecordingService }: app is in background uid UidRecord{1779cba u0a288 TPSL idle change:idle|cached procs:1 seq(0,0,0)}
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3270)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3409)
	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:83)
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2016)
	at android.os.Handler.dispatchMessage(Handler.java:107)
	at android.os.Looper.loop(Looper.java:214)
	at android.app.ActivityThread.main(ActivityThread.java:7356)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:491)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:940)
Caused by: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=de.dennisguse.opentracks/.services.TrackRecordingService }: app is in background uid UidRecord{1779cba u0a288 TPSL idle change:idle|cached procs:1 seq(0,0,0)}
	at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1616)
	at android.app.ContextImpl.startService(ContextImpl.java:1571)
	at android.content.ContextWrapper.startService(ContextWrapper.java:669)
	at android.content.ContextWrapper.startService(ContextWrapper.java:669)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.startAndBind(TrackRecordingServiceConnection.java:77)
	at de.dennisguse.opentracks.publicapi.AbstractAPIActivity.onCreate(AbstractAPIActivity.java:37)
	at android.app.Activity.performCreate(Activity.java:7824)
	at android.app.Activity.performCreate(Activity.java:7813)
	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1310)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3245)
	... 11 more
```

Haribari, 2022-08-01:
Additional hint: new function in gadgetbridge 0.68.0

dennisguse, 2022-08-01:
Duplicate of #1301.
I had i time to fix this one properly yet.
For #1301, there is a PR linked that fixes the issue in a hacky way.

closed 2022-08-01 by dennisguse