OpenTracksApp/OpenTracks #1344: Elevation data missing on stats/interval views

Cyruz143, 2022-08-03:
**Describe the bug**
Using a OnePlus Nord, it should have a barometer sensor according to the specs. Elevation data seems to be recording as it's visible in by time/by distance views but missing from interval/stats views.

Wondering if that elevation data is from GPS rather than barometer readings?

**To Reproduce**
1. Record data
2. See missing elevation data on stats/interval 
3. Swipe across to by time/by distance 
4. Elevation data seems to be visible there

If applicable:
[image]
[image]
[image]

**Technical information**
 - Device: OnePlusNord
 - OS: Oxygen OS 11.1.10.10
 - OpenTracks version: 4.0.5

dennisguse, 2022-08-04:
You can check with Trail Sense.
Most likely scenario: the phone has no such sensor.
Alternative (unlikely): the OS did not report the sensor or the sensor did not send any data.

The barometer is actually not that widely built-in.
The Fairphone's also come without.

Cyruz143, 2022-08-04:
You are correct, more digging today, no barometer... Sorry for the time waste.

closed 2022-08-04 by Cyruz143

dennisguse, 2022-08-04:
@Cyruz143 all good ;)