OpenTracksApp/OpenTracks #1350: Voice announcements - Beeps instead of voice

gitbq, 2022-08-15:
I'm a newbie to opentracks but love the concept. I am using a pixel 4 running calyxos. I have activated the voice announcements but only receive tones or beeps instead a verbalized announcement. 

The timing is about right according to my estimation, the issue is that there is no voice. 

Device. 
pixel 4 
calyxos 3.7.0
Android 12
Opentracks version 4.0.6

Are there any settings i need to tweak or anything i need to download to make this work? 

Thanks again.

dennisguse, 2022-08-15:
You need to install a TTS system. On the majority of devices this is provided by the Google Play Services.
If no TTS is available, OpenTracks creates these beeps instead (as a fallback).

closed 2022-08-15 by dennisguse

gitbq, 2022-08-17:
Thanks for the great reply. In case it helps anyone else here is what worked for me.
1. Installed RHVoice from fdroid.
2. Settings (system/language & input/text to speech input) all greyed out stay downloading language data. 
Result: still just beeps. 
3. Installed espeak from fdroid. Configured settings. Now tts settings all available. 
4. Uninstalled espeak. Espeak voice quality part average. 
5. Now success with voice announcements.