OpenTracksApp/OpenTracks #1359: Constant crashes in the background

bentolor, 2022-08-19:
Thank you very much for this great tool respecting my privacy and following FOSS values.

I'm using OpenTracks in Combination with Gadgetbridge. There i configured the right package name for OpenTracks.

Honestly I don't have a clou on the exact purpose of this integration as it never worked.

Now i regularly get this error after unlocking my lock screen.

Any ideas?

dennisguse, 2022-08-19:
Which version? 
We had some issues (to start) until 4.0.6 and not yet released is a fix to stop properly.
Problem affected Android 8+ and if OpenTracks was not already running.
I am not using this functionality, so my testing during implementation is rather limited.

dennisguse, 2022-08-19:
PS which error message?

bentolor, 2022-08-20:
Hi Dennis and thanks for your quick supply.

I created this ticket out of a crash screen and assumed it would provide more technical details right away. Not sure what tool eventually even created this ticket now. 

You're right: I'm using the F-Droid version which currently is OpenTracks  4.0.6 and Gadgetbridge 0.68.0. Also Lineage OS 18.1 / Android 11.

Will wait for an update and see if that helps.

dennisguse, 2022-08-20:
The the stop will be crashing, but start should be working (so data should be recorded).
Please reopen, if this is not true for you.

closed 2022-08-20 by dennisguse