OpenTracksApp/OpenTracks #1363: Unable to record location if app is not in active and in foreground

SuriMarks, 2022-08-23:
- [YES] Consulted [https://dontkillmyapp.com/]
- [YES] Power management: battery optimization is disabled for OpenTracks
- Device: [Samsung Galaxy S21]
- OS: [Android 12]
- OpenTracks version: [e.g. v4.0.4]

When OpenTracks is recording it only records new location data points when the app is in the foreground with the screen on. This makes it rather unusable, because as soon as the screen times out it stops tracking.
I have gone through the settings to turn off all battery optimizing restrictions as well as opened up the location permissions as much as possible, but still it does not work.

One thing I noticed it a difference in available permissions modes. Comparing these two screenshots, the maps app has a better option that is unavailable to opentracks. Not sure if this is the issue, but thought I would share it.

[image]
[image]

I can add more info if requested. Would love to get this fixed.

Thanks.

dennisguse, 2022-08-23:
The permissions are fine. OpenTracks starts a foreground service and thus only requires foreground location permission.

What did you try out already?

closed 2022-08-28 by dennisguse

dapgo, 2023-05-01:
In a previous phone with an older ver of android I don't remember issues. 
But on Android 10 with following config I had a problem during an activity.

Config during the activity:
Saving : "Medium Power Saving"
App power management:
    App that won't be put to sleep: Opentracks, OSM Dashboard

During the activity the  following events happened several times: screen off, OSM dashboard open and on foreground,  take photos. 

At the end of the activity Opentracks  have the correct total time, but  info related to location is wrong.
Total time 1h41m
Moving time 6min
Distance 0.6km

Seeing the stats, "By time" shows the x axis with the total time and correct  hearth rate. But speed or elevation are wrong and showing straight lines; probably between moments when the app went to foreground and location was received