OpenTracksApp/OpenTracks #1376: Feature: Autostart recording upon speed/cadence activity

johannesk, 2022-09-12:
**Problem:** I occasionally forget to start or stop the recording on the beginning or end of my bycicle ride. I would be much happier if I could just start riding without first fumbling on my phone.

**Suggested solution:** Automatically start recording when receiving activity from speed/cadence sensor. Stop recording after a configurable amount of time without receiving speed/cadence.

**Proposed UI**: `Settings->Bluetooth Sensors->Autostart` with new options: `Autostart recording: (on/off)` and `Stop recording after inactivity: (1s, 10s, 1m, 2m, 3m, 4m, 5m, 10m, 20m, 30m, 1h, 2h, 4h, never)`

Is this easily implementable?

Does Bluetooth LE on Android allow OpenTracks to receive activity, even when the user has not interacted with the app for hours or days?

Is anybody interested in implementing this?

dennisguse, 2022-09-12:
Sounds like a typical IFTTT problem.
Could one of those apps and use OpenTracks PublicAPI?

dennisguse, 2022-09-12:
PS the stop would be on sensor disconnect or on "no movement happened"? 
If the later then, I would move this into a new issue as it is not BLE related.

johannesk, 2022-09-12:
> PS the stop would be on sensor disconnect or on "no movement happened"? If the later then, I would move this into a new issue as it is not BLE related.

Lets take the example of a bicycle speed/cadence sensor. Once I start riding, opentracks should start recording and remember that the recording was started due to the speed/cadence sensor. When the speed/cadence sensor stops reporting data for a configured amount of time, the app can assume that I stopped riding the bicycle and therefore stop the recording. My GPS position might still be moving, since I am now walking through the office or I am on a train, but my bicycle ride has ended.

johannesk, 2022-09-12:
> Sounds like a typical IFTTT problem. Could one of those apps and use OpenTracks PublicAPI?

A quick search through the documentation of easer, tasker, and automate didn't show anything readily useable for Buetooth LE and nothing at all for speed/cadence sensors. Since this is for sensors which are already reporting data to OpenTracks, my guess is that this feature fits well into OpenTracks.

dennisguse, 2022-09-13:
Wow - I am mindblown.

So, the problem is huge. 
It starts with: BLE devices (usually) do not establish a connection when turned on.
They rather wait for incoming connection.
Usually they do not even know to whom to connect to. 
They just announced their presence.  

This means the phone needs to wait until a device gets present (ie. it receives a announce packet).
And now the fun part: how does an app tell Android that it should start the app, when a certain devices is close by (aka receiving announcements)?

OpenTracks is starting a Foreground service for recording and also BLE things. And here we tell Android to do a BLE device discovery for certain device classes.
The actual fun part: how to do this properly in the background?

dennisguse, 2022-09-13:
Okay, so there is the CompanionDevice (Android 8.0+): https://developer.android.com/guide/topics/connectivity/companion-device-pairing