OpenTracksApp/OpenTracks #1377: Icon set via PublicAPI isn't used in track details

arjan-s, 2022-09-13:
**Describe the bug**
When a track icon is set via the PublicAPI, it is only used in the list of all tracks. When tapping on the track it opens the track details, and instead of the correct icon, it shows the generic icon. When setting the category/icon manually, it shows the icon in both screens correctly. 

**Technical information**
 - Device: Samsung Galaxy Fold3 
 - OS: Samsung OneUI 4.1 (android 12)
 - OpenTracks version: 4.0.8

dennisguse, 2022-09-13:
Thx. I can confirm this by trying to use a different category than icon (via edit functionality).

PS/ IIRC I am not even sure that we export the icon to KML/GPX (and also not import).
So, this feature slightly broken anyhow ;)

dennisguse, 2022-09-13:
Note: icon export/import is NOT supported by our GPX implementation.
KML should work fine.

dennisguse, 2023-06-27:
This is hopefully fixed by #1608.
If not, please re-open.

closed 2023-06-27 by dennisguse