OpenTracksApp/OpenTracks #1382: please add “android.permission.ACCESS_BACKGROUND_LOCATION”

SunDaSora, 2022-09-24:
on “miui”，you need “android.permission.ACCESS_BACKGROUND_LOCATION” to keep it work。
 Detail：在小米手机上，没有这项权限的话，会导致关闭屏幕后应用无法获取GPS位置。由于其权限管理组件只会给请求了此项权限的应用添加“始终允许”的选项，而没有请求此权限的应用只能在前台使用GPS，并且无法手动更改。

dennisguse, 2022-09-27:
We should not need this permission (due to the use of a ForegroundService).
Can you provide a link to their documentation?
Or were you found this suggestion?

closed 2022-09-27 by dennisguse

SunDaSora, 2022-10-04:
（My English is not very good. If something confuses you, please ignore it.）
OK，you are right.I found what happend.On MIUI you need the Permission to use GPS when app background(The system will consider your app to be in the background after closing the screen.)，it named “省电策略”，“无限制” 。 I think it also named “Battery Optimization”。There is 2 Picture，one is your app tell me noGPS,the other is “省电策略”. it sad “It can only identify the application state and take restrictive measures to restrict background networking, positioning,  sensor access and CPU usage。”I changed it from smart limit to unlimited, and the application works normally.
[image]
[image]

dennisguse, 2022-10-05:
Thanks for the feedback - it is on the Todo list.

PS don't worry about your English.
You can just your native language (and also post an automatic translation).
That's sufficient.
And the worst case, we ask questions (hopefully simple) or somebody will be able to translate.