OpenTracksApp/OpenTracks #1383: app being blocked by play protect

mneko, 2022-09-26:
**Describe the bug**
I'm trying to install from fdroid from the official repo. It pops up with a "This app can collect data that may be used to track you" message and blocks the install.

**Technical information**
 Samsung Galaxy Note 10+
Android 12
Occurs with opentracks versions 4.0.7 thru 4.1.0

According to https://stackoverflow.com/questions/49092689/app-installation-blocked-with-a-pop-up-saying-app-may-collect-data-to-track-user you might just have an out of date ssl certificate or something of the sort

dennisguse, 2022-09-27:
Can you provide a screenshot?

We are not using any network...

dennisguse, 2022-09-27:
If you saw this message: [image]

... this happens every now and then.
Google PlayServices "virus/malware scanner" is not really non-PlayStore friendly.
Nothing we can do about.

nourkagha, 2022-10-01:
Same issue here. Here's a screenshot. [image]

dennisguse, 2022-10-03:
If you enable PlayProtect, you have to live with it's features/ malfunctions.

Nothing we can do about.
You may check with the Fdroid people (but I guess they have the same problem).

Sorry.

closed 2022-10-03 by dennisguse