OpenTracksApp/OpenTracks #1384: KarooOS: Export with no compatible file manager

neurer, 2022-09-26:
**Is your feature request related to a problem? Please describe.**
Unable to set track export directory when "No compatible file manager installed". Instant post-workout export is unavailable respectively.

**Describe the solution you'd like**
To be able to set export directory either manually, via config or to be able to pick a file manager. [Amaze](https://teamamaze.xyz/) seems to fit the spirit of things, if that's something that could be checked against.

**Describe alternatives you've considered**
Tried different (Android 8.1 Stock) file managers with different signatures; no luck. Tried file managers available from f-droid; no luck. Manual export via share dialogue works, but does seem tedious in the long run. I'm aware of #1334.

**Additional context**
I've been using OpenTracks with the (Android 8.1 based) KarooOS; on the Hammerhead Karoo 2 cycling computer. Works like a charm. But there's no file manager included, though sideloading anything works without issues. I'm aware that this is mainly a vendor/policy issue, but it sure does sound like a smart thing to have a fully featured OpenTracks running in some shape or form on this (decently) open platform.

Any advise and/or solution would be much appreciated. Thanks for your time, the exceptional work and your consideration.

dennisguse, 2022-09-27:
We are using ScopedStorage (as required by Android 11+): https://developer.android.com/about/versions/11/privacy/storage

OpenTracks actually requests that the user selects a folder and that it gets write permission for this folder.
This happens here: https://github.com/OpenTracksApp/OpenTracks/blob/2b530a1c53776049f1da109e97f346d5e278f068/src/main/java/de/dennisguse/opentracks/io/file/importer/DirectoryChooserActivity.java

So, we don't need a filemanager - we actually need an application that can handle that intent. I assume that this is the OS + a filemanager.
So, I don't even know if a filemanager that is sideloaded would allow this...

Usually I would recommend to complain to the OS vendor :)

dennisguse, 2022-09-27:
PS/ I just checked their webpage.
The hardware looks good (although I am not a fan o special purpose devices).
Still that's a pretty decent device and at least the display is way more suiting than my re-purposed Nexus 4 ;)
I would recommend to just approach them and tell them about the problem.

In OpenTracks it would be possible (but requires some quite some work) to use the old WRITE_EXTERNAL_STORAGE permission + folder selection: https://developer.android.com/training/data-storage

And is OpenTracks' elevation gain working on the Karoo?

neurer, 2022-09-27:
> And is OpenTracks' elevation gain working on the Karoo?

Yes. Yes it does. The HRM that came with it does too :partying_face: 

> Still that's a pretty decent device and at least the display is way more suiting than my re-purposed Nexus 4 ;)

That's what I thought and so I went for it, replacing my re-purposed XZ1 Compact. I quite like everything about their Android-based approach and the possibilities that come with it. A new eco system in very early stages; which recently got bought by SRAM though. Still, it's a well engineered and sensible solution to keeping data from getting locked into any of the vendor eco systems. No GAPPS. No Play Services integration. Works (even though in a limited fashion) without any account. I run Organic Maps for navigation on it.

While I do understand your suggestion to just ask them, I'm not sure what a smart or workable angle would be. They provide hardware, infrastructure and tools for me to feed them with data. And I'd like to run OpenTracks. Looking at KarooOS in its current incarnation, one could draw the conclusion that it's current "openness" is something that just hasn't been locked away yet.

Exciting times.

PS. Since we appear be in the same city. If you need the device for dev/testing, please do let me know.

dennisguse, 2022-09-29:
@neurer for offline communication: https://chaos.social/web/@dennisguse

dennisguse, 2023-06-07:
Sufficiently fixed by #1318

closed 2023-06-07 by dennisguse