OpenTracksApp/OpenTracks #1387: Tracking issue.

JahidFariz, 2022-09-30:
This app tracks well. But sometimes Android os kills this app while running in the background. And this app tracks even if the GPS signal is lost. It won't show any alert message in the background the signal is lost. 

Add a feature to disable battery optimization. 

Make a change in location permission while using the app to allow all the time.

dennisguse, 2022-10-03:
You are right: disabling battery optimization is an improvement.

About the background location permission: that is not needed as we use a foreground service while tracking.

SunDaSora, 2022-10-04:
This is very useful, I have the same problem, I use MIUI, if you do not change the application to unlimited, this will happen.