OpenTracksApp/OpenTracks #1388: Exported kmz is invalid

spaetz, 2022-10-05:
Dear all, thanks for OpenTracks. We have v 4.10 from fdroid installed. Attached a doc.kml file that is inside the exported kmz file. It fails to display anything in Google maps, Arcgis, Gnome maps or other applications that can display kml/kmz files.

The kml file only has the latitude/longitude replaced with different values for privacy reasons.

I have not found a KML validator tool, so I am a bit at a loss as to why those tools would not be able to view the file.

spaetz, 2022-10-05:
[doc.kml.txt](https://github.com/OpenTracksApp/OpenTracks/files/9713511/doc.kml.txt)
kml file with .txt suffix to enable uploading.

dennisguse, 2022-10-05:
Looks good; not your fault ;)
KML 2.2 and 2.3 are very similar (2.3 integrates some Google extensions everybody was using).
Sadly, 2.3 is not as well established as it should be...

KML is more meant for backup and restore within OpenTracks (e.g. it lacks standard data fields for cadence). GPX is way nicer (but does not support 1. - see below) - so it is not that cool for backup and restore.

TL;DR

OpenTracks does two things that are in line with the standard but often not well handled.
1. Export of track points without location aka empty `<coord />`; often these are interpreted as 0,0.
2. Timestamps comply to xsd:datetime and contain an offset (aka offset to UTC at recording location)
We did these two things already in KML2.2 though.

This works for OsmAnd (they just added support for KML 2.3) and everything based on OpenLayers.
Also GPSVisualizer works https://www.gpsvisualizer.com/ (I reported it some weeks ago and they fixed it within hours).

Regarding Gnome Maps: can you submit a bug report to them?

Same thing for all other tools that are not working nicely ;)

PS most tools only support KML not KMZ :/

PS/ KML 2.3 is not supported by Google Maps. Don't care why.

spaetz, 2022-10-05:
Thanks for the quick reply. Will Check osmand, had not considered that yet. And will file a report against gnome maps nudging them a little. Thank you for the detailed explanations, appreciated! I'll be closing this issue as there is nothing on the OpenTracks side that should be done...