OpenTracksApp/OpenTracks #1393: Setting to allow user to change the marker behavior

PhymIsBored, 2022-10-11:
When you place a marker while tracking the marker is only placed when you finish editing the marker. 
Instead allow the user to decide if the marker should be placed when the "add marker"-button is pressed or when the "Add"-button is pressed.
For example on the GPS settings page.

dennisguse, 2022-10-11:
What is the actual problem?
That you need press two buttons?
That you need the market is created too late (because we get the coordinates when save is pressed)?

For your suggested solution: what happens if you press cancel?

PhymIsBored, 2022-10-11:
When you want to palace a marker you need to stop and stand at the spot that you want to mark until you press the add button. 
If I walk by a point that I want to mark, and I press the "add marker"-button, it does not place the marker at the point I was at when I pressed the button, instead it places it at the coordinates I am at when I press the "add"-button on the screen that allows you to edit the marker you are going to place.
The issue is that I have to stop for as long as it takes me to input the information I want the marker to have (Description, etc.).
Intuitively I expected it to save the GPS-coordinates of the location that I initially pressed the marker-symbol.

dennisguse, 2022-10-12:
Sounds reasonable.
Let's just change the behavior to fetch coordinates inmediatamente rather than on save. 
The later seems unintuitive.

For now, you can save the marker immediately and then edit it later. That's the only option to store the current location immediately.

commit 9ed1180b (2024-01-12): New marker get current location (on create rather than on save).

Fixes #1393.
commit 541817a2 (2024-01-12): New marker get current location (on create rather than on save).

Fixes #1393.
PR #1825 (2024-01-12): New marker get current location (on create rather than on save).
Fixes #1393.
commit b1fafc4f (2024-01-12): New marker get current location (on create rather than on save).

Fixes #1393.
closed 2024-01-12 by dennisguse