OpenTracksApp/OpenTracks #1396: Migrate weblate

dennisguse, 2022-10-14:
From https://weblate.bubu1.eu to https://translate.codeberg.org/

dennisguse, 2022-10-14:
@pstorch FYI

comradekingu, 2022-10-17:
@dennisguse
I implore you not to use Codeberg on account of their stance against freedom https://codeberg.org/Codeberg/org/src/branch/main/TermsOfUse.md
This rug-pull operation was neither mentioned up front, or to users later.
After that there is no tangible reason to use Codeberg, as that was its one and only premise to operate by.

https://hosted.weblate.org OTOH has a great record in the service of freedom, and is maintained by Weblate staff.
Its community is also greater.
Its services are on offer for libre projects via https://hosted.weblate.org/hosting

dennisguse, 2022-10-17:
@comradekingu I am open for discussion (and dont care too much where to host the translations as I am not using these systems that often(.

I just went briefly through Codeberg's terms and conditions and have not seen any obvious red flags.

Can you elaborate in the problems that you are seeing?

@pstorch @mondlicht-und-sterne @rgmf FYI

pstorch, 2022-10-17:
@comradekingu what Do you mean by "tug pull operation"? I only remember the discussion as codeberg clarified the usage of their service to  FOSS projects and they don't allow a misuse as private cloud storage, which is a legit move IMHO. 
Is there anything else you have in mind?

rgmf, 2022-10-17:
I didn't read carefully the terms and conditions of Codeberg but I thought it was a great service for libre software. It's a sorprised for me reading complaints about it.

mondlicht-und-sterne, 2022-10-17:
Oh kingu, he hATe me.  Because I had translated another project into norwegian, I got angry emails from him all the time.
Maybe it's enough if I don't translate this language for OpenTracks.

dennisguse, 2022-10-18:
@comradekingu what about hosting the translation on both weblate instances?
We can just configure both to automatically create pull requests and thus may avoid most conflicts due to diverging changes.

@mondlicht-und-sterne I guess it is not hate, but rather a different perspective on how to solve a problem. And then frustration and communication  is spiraling downwards.

comradekingu, 2022-10-21:
@dennisguse 
Why would you want anything hosted on the Codeberg Weblate I wonder?
There is no upside there, and it makes communication between translators worse. I rely on that.

dennisguse, 2022-11-24:
As we had no real discussion here, I flipped a coin.
`Weblate.org` won.

closed 2022-11-24 by dennisguse

Bubu, 2023-03-14:
Your f-droid metadata still points to weblate.bubu1.eu: https://gitlab.com/fdroid/fdroiddata/-/blob/master/metadata/de.dennisguse.opentracks.yml#L8

dennisguse, 2023-03-14:
@Bubu thanks for the reminder :)

https://gitlab.com/fdroid/fdroiddata/-/merge_requests/12748

commit 48aaca13 (2022-11-24): Migrated to hosted.weblate.org

Reason: weblate.bubu1.eu shutdown end of 2022.
Alternative: translate.codeberg.org
Selection method: flipped a coin

Fixes #1396.