OpenTracksApp/OpenTracks #1421: GPS is not aktivated - OpenTracks is recording without GPS signal

marsch1975, 2022-11-15:
Hi,

every now and then OpenTracks tells me "GPS is not aktiv - OpenTraks is recording without GPS data" but at that moment the line of the journey is fine, in some other cases the line is broken...

I set the battery settings to not interupt OpenTracks which is working fine with other apps...

Samsung Galaxy S21 5G
Android 12
One UI 4.1
OpenTracks 4.1.2

What else can I do?

Cheers
Markus

dennisguse, 2022-11-15:
Go to someplace with a better GPS reception :)

For real: not that much (if it is not a battery issue).
I would need to check the code: but this message is popping up after 30s(?) if there was no location received in between.
It is just user feedback and has no other impact.

However, two consecutive locations are too far apart (configurable; default 50m) then a new segment is created (aka the line is broken).

closed 2022-11-15 by dennisguse

marsch1975, 2022-11-17:
Hi Dennis,

thanks for your reply.

In fakt, I'm under the clear blue sky without high buildings around when it happens..

Most of the times it's just missing for serveral Meters up to some 100Meters i guess and once I was starting at home and after about 10 to 15 minutes it started recording, that was about 4km...

Time in s "smallest"
Recording is 10m (recommended)...
Max distance is 200m (recommended)
GPS 50m (recommended)

So you want me to change the GPS of 50m? Higher or lower?

marsch1975, 2022-11-28:
How do I make OpenTracks show coordinates? I've been the menu up and down... can't find it...

dennisguse, 2022-12-01:
Settings -> User Interface -> Customize recording Layout -> (choose one or create a new one) -> add the one with Latitude/Longitude

marsch1975, 2022-12-12:
Oh, ok, it wasn't easy to find that I can configure the screen myself. When you go on "customize/create your own" it looks like threre is nothing further "like it's empty feature"...

It helped to start the recordings just a few seconds before starting out...

Thanks for all your help!