OpenTracksApp/OpenTracks #1422: Open Tracks crashed while recording a walk

ThaBlindape, 2022-11-16:
### App information
* ID: de.dennisguse.opentracks
* Version: 5004 v4.0.7

### Device information
* Brand: samsung
* Device: crownlte
* Model: SM-N960F
* Id: QP1A.190711.020
* Product: crownltexx

### Firmware
* SDK: 29
* Release: 10
* Incremental: N960FXXU9FVG2

### Cause of error
```java
Exception in thread "TrackDataHub": java.lang.RuntimeException: stopTime cannot be less than startTime: 2022-11-16T05:18:15.279Z 2022-11-16T05:12:09.305Z
	at de.dennisguse.opentracks.stats.TrackStatistics.setStopTime(TrackStatistics.java:196)
	at de.dennisguse.opentracks.stats.TrackStatisticsUpdater.addTrackPoint(TrackStatisticsUpdater.java:108)
	at de.dennisguse.opentracks.data.TrackDataHub.notifyTrackPointsTableUpdate(TrackDataHub.java:362)
	at de.dennisguse.opentracks.data.TrackDataHub.loadDataForListener(TrackDataHub.java:242)
	at de.dennisguse.opentracks.data.TrackDataHub.lambda$registerTrackDataListener$1$de-dennisguse-opentracks-data-TrackDataHub(TrackDataHub.java:186)
	at de.dennisguse.opentracks.data.TrackDataHub$$ExternalSyntheticLambda0.run(Unknown Source:4)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:237)
	at android.os.HandlerThread.run(HandlerThread.java:67)
```
Hi,

I had started recording a walk i was on. After about 1 km the app crashed.

cheers,
john

dennisguse, 2022-11-16:
Do you any more insight?
* Was the screen on?
* Did you use any BLE sensors?

What happened: you traveled back in time (aka locations were processed not in their temporal order).
Alternative option: your phone's clock jumped back, but OpenTracks should cover that (see `MonotonicClock`).

Anyhow: the only option that comes to my mind is that the processing order was changed (aka a race condition).
This _might_ be fixed in v4.1.2 as the GPS coordinates are then processed in the same thread as all other events (and thus keep the order).

dennisguse, 2023-04-23:
No more infos provided.

closed 2023-04-23 by dennisguse