OpenTracksApp/OpenTracks #1424: Bluetooth pressure sensor

vlszabolcs, 2022-11-17:
#[continue here:](https://github.com/OpenTracksApp/OpenTracks/discussions/1423) 
I would like to use an Arduino Nano 33 BLE Sense dev board as an external Bluetooth pressure sensor (my phone hasn't built in) to get more precise altitude data.

What is currently working:
I created a code  for the nano33 sense what can you[ find here.](https://github.com/vlszabolcs/ble_sensors) I use Environmental Sensing service (0x181A) and Pressure characteristic 
(0x2A6D, datatype: uint32). (Notice: For testing  BLE characteristic i use the Elevation, Humidity,Temperature characteristic and an unknown characteristic for setup the sea level pressure to corrigation the altitude.)
I tested in nRF Connect and looks like it working there. I checked the ble sensor pressure value another device and it is correct. I think this service and characteristic setup can use with any bluetooth device for mesauring pressure . 

The problem for me i'm not really know the java. In the sensor settings I created a new generic option to the scan the ble pressure device but after i lost in the code. So i need some help to implement this if you think it s usefull.

[image]

dennisguse, 2022-11-17:
Okay - that looks like something is already working.

Would you like to fetch the elevation or the pressure data?
For practical reasons it might be easier to fetch the pressure data as in OpenTracks we have the computation of the gain/loss already in place - it is "just" fetched from the internal sensors.
Steps to be taken:
- [ ] we need some test data to implement the decoding (see BluetoothUtilsTest)
- [ ] add sensor selection to be settings (`PreferencesUtils` and `BluetoothUtils`: service, characteristics, and measurement UUID)
- [ ] add data decoding to BlueoothUtils (should be similar to the heartbeat code)
- [ ] do some magic to create use `AltitudeChange` (for BLE data: we can use SensorDataSet)
- [ ] figure out what we do if an internal pressure sensor and a BLE pressure sensor are connected.

PS/ all doable no rocket science included.
@vlszabolcs I can help you down the road ;)
Can you already compile and start OpenTracks on your phone?
If not that needed before everything: just get AndroidStudio and have some fun :)

vlszabolcs, 2022-11-21:
I would like to use only the pressure data. The other data was used for testing ble characteristics but I updated the arduino code to use only the pressure.

As you writed,  my idea was also that, the bluetooth device provide only local pressure data and then OpenTracks processing it (as now internal pressure sensor data).

I succesfully compiled and running on my test phone (Xiaomi Redmi 4X). I tried to do some modifications too. So now the app can connect to the BLE pressure device, reading and giving back the data in hPa. I think the next step is to somehow use the data for altitude calculation. [Here](https://github.com/vlszabolcs/OpenTracks/tree/BLE_barometric_pressure_sensor) is the code.   

[Here ](https://github.com/vlszabolcs/ble_sensors/blob/master/test/Log%202022-11-21%2015_34_42.txt)some test data form NRF conncet.

Thanks for your instruction!

dennisguse, 2022-11-22:
@vlszabolcs Can you create a Pull Request from your fork?

vlszabolcs, 2022-11-22:
I made some mistakes in the _sensorData_ and _sensors_ folders files. After I created e.g. the SensorDataBluetoothPressure, I got a problem e.g. in the SensorDataBluetootHeartRate:

>Class 'SensorDataHeartRate' must either be declared abstract or implement abstract method 'getNoneValues()' in 'SensorData'

For solve the problem i used Quick-fixes but this generated other modifications. I haven't found any other solution yet.

vlszabolcs, 2023-01-04:
I got unplanned work. I am going to continue this project in the middle of February.

Some updates from the BLE sensor device: I found this [BLE sensor tag](https://www.aliexpress.com/item/32859423925.html?spm=a2g0o.cart.0.0.195338daXE5gyL&mp=1). From the picture and description it looks like this device is ready to use in terms of hardware . I ordered two pieces and hope it will be in reality.

dennisguse, 2023-01-04:
I almost did the same: I ordered a RuuviTag: https://ruuvi.com/ruuvitag/
Hardware is complete (incl. case) and the firmware is open-source, but it does not implement `Environmental Sensing service`.
That is a todo...

vlszabolcs, 2023-02-01:
This RuuviTag is an excellent device for the task. 
As I saw, the RuuviTag uses the `Nordic UART service` which can also be used to extract the necessary data. I think for the first attempt, we could use the original firmware with the UART service  to accomplish the next steps. It if necessary in the future easily modify.

dennisguse, 2023-02-02:
@vlszabolcs not really; the Ruuvi-Tag broadcasts (via a custom/proprietary) protocol.
First of all, this is not privacy-friendly as you basically carry personal public hotspot.
Second, they are the only ones using this protocol.
And implementing a non-standard protocol into OpenTracks is (from my perspective) out of scope.
Third, there is an existing standard.

And finally, the implementation to use broadcasted data might be very different than the current connection-based one's in OpenTracks.

dennisguse, 2023-06-08:
What is missing        
* [ ] implement merged sensor driver for internal and BLE pressure sensor
* [ ] add settings UI (get from #1433 commit 534b6c7e3a8a4e9b982ea2d1e1717defeb5f2576 )

dennisguse, 2023-12-30:
@vlszabolcs We now got support for this.
Do you have the time to test it?

vlszabolcs, 2024-01-01:
@dennisguse This is great, Yes, I will test it. I will apply a method similar to what is described in the ['README_TESTED_SENSORS.md'](https://github.com/OpenTracksApp/OpenTracks/blob/main/README_TESTED_SENSORS.md#tested-barometric-sensor-smartphones) for the testing. Is that appropriate?

dennisguse, 2024-01-01:
That sounds good.

PS I had no sensor at hand as I have not yet found a shop where to buy one that is actually providing Environmental Sensing. 

So, you will be the first real user :)

vlszabolcs, 2024-01-03:
I tested a little bit and i think it's very cool!!

I used version [5603 v4.10.0 Nightly](https://github.com/OpenTracksApp/OpenTracks/releases/tag/v4.10.0). During testing, I frequently encountered an issue similar to the one in [1799](https://github.com/OpenTracksApp/OpenTracks/issues/1799#issue-2058724787).

The BLE device successfully connected and provided data to the application using the specified service and characteristic. The Gain and Loss remain at 0 for a while, but there seems to be an issue with data processing.

As per the [Bluetooth Characteristic Pressure ](https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.characteristic.pressure.xml) description, the BLE device provides Pa values accurate to 0.1. However, in the application's Logcat, it shows "VALUE: AtmosphericPressure[value=9970140.0]".

For instance:
 
* The measured atmospheric pressure value: 997.014 hPa = 99701.4 Pa
* Sent via BLE in uint32 format: 997014
* Parsed value in nRF Connect: 99701.4 Pa
* OpenTracks Logcat: "VALUE: AtmosphericPressure[value=9970140.0]"

If I understand correctly, the [`ofPA()`](https://github.com/OpenTracksApp/OpenTracks/blob/c6b4e738b9b609d732ba397c352df23d2a3770cb/src/main/java/de/dennisguse/opentracks/data/models/AtmosphericPressure.java#L5C42-L5C42) function  expects values in  hPa, but now it multiplies the Pascal unit by 100, creating centi(?)Pascal values. 

As a quick test, i changed form `ofPA()` to `ofHPA()`  in the [BluetoothHandlerBarometricPressure](https://github.com/OpenTracksApp/OpenTracks/blob/c6b4e738b9b609d732ba397c352df23d2a3770cb/src/main/java/de/dennisguse/opentracks/sensors/BluetoothHandlerBarometricPressure.java#L52) and get correct value for example: "VALUE: AtmosphericPressure[value=99701.1]". With this  modification, the [Sensor Test](https://github.com/OpenTracksApp/OpenTracks/blob/main/README_TESTED_SENSORS.md#tested-barometric-sensor-smartphones)  first three elements **worked**.

dennisguse, 2024-01-04:
That's sounds pretty good :D
Can you create a pull request for this fix?

PS/ Can you also check if #1810 fixes the problem of #1799?

vlszabolcs, 2024-01-05:
Yes, I can confirm that  #1810 fixes the problem! 

Thanks for the pull request! #1813 now displays data in hPa units in the logcat! Great! :D 

I'll soon proceed with the fourth test.

One more observation: sometimes, too many decimal places appear in the logcat. For example:
logcat: 1001.60297 hPa
BLE device: 1001.602 hPa

I assume this won't cause significant differences, but I think it might be worthwhile to implement rounding or cut somewhere.
Considering that most affordable atmospheric pressure sensors can measure with an accuracy of 0.00 hPa, it might be beneficial to round according to this precision.

dennisguse, 2024-01-05:
Thanks for testing.
I will create a new release now as there are now some things less broken :D

PS/ I will refactor the AtmosphericPressure to store data in Pa (as int) instead of hPa (as float).
Then this rounding stuff becomes obsolete :)

closed 2024-01-05 by dennisguse

dennisguse, 2024-01-05:
@vlszabolcs I was looking for a compatible BLE sensor and nope.
I have found none that is commercially available.
SensorPush and RuuviTag look nice, but neither supports Environmental Sensing Service.

Any ideas?

PS/ I spent hours on searching already...

PPS/ So, I have no sensor for testing :D

vlszabolcs, 2024-01-06:
Nightly v4.10.1 works great !

I haven't found them yet either. I've tested with dev boards like [Arduino Nano 33 BLE Sense](https://store.arduino.cc/products/arduino-nano-33-ble-sense?selectedStore=eu) and [nRF51 Sensor Tag](https://www.aliexpress.com/item/32859423925.html?spm=a2g0o.detail.0.0.7168WDSNWDSN1O&gps-id=pcDetailTopMoreOtherSeller&scm=1007.40050.362094.0&scm_id=1007.40050.362094.0&scm-url=1007.40050.362094.0&pvid=65c4103d-486d-4559-8ead-f12c1cbb900e&_t=gps-id%3ApcDetailTopMoreOtherSeller%2Cscm-url%3A1007.40050.362094.0%2Cpvid%3A65c4103d-486d-4559-8ead-f12c1cbb900e%2Ctpp_buckets%3A668%232846%238115%232000&pdp_npi=4%40dis%21HUF%217282.40%215970.93%21%21%2120.49%2116.80%21%402103200617045371396292353e90eb%2165326366709%21rec%21HU%21874335205%21). It's possible to further develop these too, but I believe there's great potential in the Ruuvi Tag for a first attempt. It's fully developed hardware-wise, and basic functions work perfectly (e.g., power management, connectivity, sensor management, error handling).

Since it's open source, it might be easier to create a modified firmware with environmental services. As far as I know, OTA is possible, enabling an average user to update the tag. I'm planning to reach out to the Ruuvi Tag team to explore the possibilities regarding this.

dennisguse, 2024-01-08:
That's awesome :D
That's really cool!

What I did so far:
1. [somewhen last year] Ordered a RuuviTag (sadly only supports broadcast)
2. ... I contact customer support about ESS and their response was "it's open-source"
3. [Dec/23] contacted SensorPush and asked for ESS
4. .. they forwarded the request to their engineering team ;)

PS/ The RuuviTag is relatively large and the new RuuviTag Pro is even larger.
At least they don't feel like something I want to attach to my bike.

Alternatives I've considered:
* BangleJS 2 has a barometer and is waterproof 
* Using an older Android phone (got an S4 with baromter)
* Nicla Sense (small, expansive, no battery support)
* Building stuff myself (never done, so yeah)

Regarding nRF SensorTag: are those programmable?

vlszabolcs, 2024-01-09:
Programming the nRF Sensor Tag involves a slightly more complex task of flashing and debugging compared to a BLE Nano 33 or a Nicla Sense because it lacks a built-in programmer. 

The RuuviTag can be developed with a similar level of complexity as the nRF Sensor Tag. Regarding its size, I initially considered it from a hiking perspective, but indeed, it would be uncomfortable for cycling.

I've looked into the Nicla Sense before, but due to the many unused sensors, it can be relatively expensive.

Overall, the good news is that there are available devices and solutions for the ESS, but each requires some degree of development. For my initial attempt, I'm trying to develop the Ruuvi Tag by adding ESS functionality to it.

dennisguse, 2024-01-12:
@vlszabolcs can we move this discussion into a new issue - this might be kind of ongoing and well hidden as this one is closed :D

PS/ For now, I will do other things and come back to this hardware topic somewhen in March/April :D

commit f369c4fd (2023-12-28): Implement gain/loss via Bluetooth LE Environmental Sensing Sensor.

Fixes #1424.
commit 8a1dc467 (2023-12-28): Implement gain/loss via Bluetooth LE Environmental Sensing Sensor.

Fixes #1424.
commit a5a44c68 (2023-12-29): Implement gain/loss via Bluetooth LE Environmental Sensing Sensor.

Fixes #1424.
commit 10fe8158 (2023-12-30): Implement gain/loss via Bluetooth LE Environmental Sensing Sensor.

Fixes #1424.