OpenTracksApp/OpenTracks #1425: App stops recording on its own

hollunder, 2022-11-20:
**Describe the bug**
The other day I was cycling about 1 1/2 hours one direction and intended to use OpenTracks to record the trip. Unfortunately in both directions OpenTracks stopped recording on its own. On the way there after 37:58, on the way back after 11:27. I did occasionally stop and fumble with the phone because I was not familiar with the route and there were some redirects but I definitely did not stop the recording. Afterwards OpenTracks was still showing in the notification area and the only way to stop it that I found was via "force stop".

**To Reproduce**
It happens at seemingly random intervals during regular use.

If applicable:
* add screenshots
* add logcat output

**Technical information**
 - Device: Motorola Moto G5 (2017)
 - OS: LineageOS 18.1
 - OpenTracks version: v 4.1.2 installed via F-Droid

dennisguse, 2022-11-20:
Not sure how this could happen.
Sounds like something crashed (seemingly in the background).

Is this new or did it happen with prior versions (pre 4.1.2)?
There was a change that might explain it (GPS locations in the same thread as everything else in TrackRecordingService).

hollunder, 2022-11-21:
I am not entirely sure as I haven't used the app that much.
I do have a 5:56:35 long track from August 27, a 3:09:05 long from September 18, a 38:36 and a 41:36 from October 9. They were all fine, but the October ones were short enough that I could have just been lucky.

I usually have my apps up-to-date, so those were most likely recorded with the version available in F-Droid at the time. So August and September were < 4.1.0, October was almost certainly with 4.1.1.

dennisguse, 2022-11-22:
Of you had the time to use it more often that would be great. 
I am not sure what is happening, bit it seems to be a race condition and it is happening relatively often. 

And a logcat after it happened would be amazing.

hollunder, 2022-11-23:
I have no clue how to get at logs on Android. Can you give me a pointer to some good guide?

dennisguse, 2022-11-24:
Simplest thing is usually with AndroidStudio: https://www.youtube.com/watch?v=smOzLSTaso0
This video looks okay: https://www.youtube.com/watch?v=smOzLSTaso0

It has actually been ages since I did this the first time, so I am not really good guide for this.

When you have the logs, you can search for `Exception` and there should be some error message with a lot of lines showing where it happened.
And be aware that logs may contain private information, so be sure to not upload anything online that you wouldn't like to share.

dennisguse, 2023-04-25:
@hollunder any more infos?

hollunder, 2023-04-25:
Not yet. I haven't been on long hikes recently. I am using the app a little bit these days, but for much shorter tracks.

dennisguse, 2023-04-27:
Let's hope this is fixed then :)
Please re-open if it occurs again.

closed 2023-04-27 by dennisguse