OpenTracksApp/OpenTracks #1426: Re-implement StatisticsRecordedFragment's sensor statistics UI

dennisguse, 2022-11-22:
**Is your feature request related to a problem? Please describe.**
While refactoring #1386, I removed the SensorStatistics (e.g., max heartrate) from the StatisticsRecordedFragment.
It was a side effect of reducing complexity to add more custom UIs to the StatisticsRecordingFragment.

**Describe the solution you'd like**
Re-implement a RecyclerView.Adapter that manages the SensorStatistics on the UI.

**Additional context**
This seems like a separate UI, so re-using existing components might not be beneficial.
Layout: `stats_generic_item.xml`

c33s, 2023-05-06:
would this also include having pace statistics (max pace, min pace, avg pace) in the recorded track detail view (currently v4.1.1 only visible while recording)?

dennisguse, 2023-05-06:
@c33s yes. 
I needed to remove this due to some refactoring - some things were tied together too much to implement the heart rate zone feature. 

And I had no time to reimplement it.
Help is appreciated.

c33s, 2023-05-27:
please provide the 4.1.2 here and on fdroid as long as this feature isn't reimplemented. currenlty it is not possible to downgrade from a trusted source :(

MrS1R1US, 2023-05-28:
[OpenTracks.v4.1.2.zip](https://github.com/OpenTracksApp/OpenTracks/files/11583712/OpenTracks.v4.1.2.zip) I use this version because I need heart rate data for comparison (

c33s, 2023-05-28:
> [OpenTracks.v4.1.2.zip](https://github.com/OpenTracksApp/OpenTracks/files/11583712/OpenTracks.v4.1.2.zip) I use this version because I need heart rate data for comparison (

thank you very much. as far as i know this is the playstore version. i try to stick by the fdroid versions but it is gone from fdroid:

https://f-droid.org/en/packages/de.dennisguse.opentracks/

[image]

i assume it is because of the fact that no direct downgrade is possible via fdroid.

for the records 
- i copied the 4.1.2 apk from a 2nd phone to the main phone already having installed a 4.3 version. 
- i exported the data from 4.3
- the installation of 4.1.2 failed
- uninstalled 4.3
- installed 4.1.2 (worked)
- imported the data 
- i got back the average and the max heart rates even from tracks i recorded with 4.3

for me a key feature got lost with 4.2. was it really worth it (the refactoring) to remove such a valuable key feature?

@dennisguse how much would it cost to sponsor this feature? maybe we can start a collection.

MrS1R1US, 2023-05-28:
> thank you very much. as far as i know this is the playstore version. i try to stick by the fdroid versions but it is gone from fdroid:
>
> https://f-droid.org/en/packages/de.dennisguse.opentracks/ [...]

Dennisguse dropped this version for me. Tell me, how is it different from the f-droid version?

dennisguse, 2023-05-28:
@MrS1R1US it is a different packageId and it was compiled by me (rather than the F-Droid people).
Functionality is the same and you trust my computer that compiled it rather than F-Droid's computer.
@c33s I cannot build the F-Droid version as the signing is happening by F-Droid alone.
:shrug: 

Regarding the removal.
The code was well to complicated to be changed and also would be hard to change for new joiners.
So, I decided to rewrite the important parts and implemented a new feature.
Even that took me quite a while aka well too long.

What is actually missing.
The StatisticsRecordedFragment lacks the UI code to show the sensor-based statistics.
All computations are still in place.

What next?
Basically anybody can take up this work and implement this.
It is not rocket science.

I lack quite some spare time (aka I have better things to do and we have summer).
Also, I would rather do some things that I need personally (like #1423).

@c33s this actually boils down to motivation/ resources.
You can try to either do it yourself or motivate people to do the work or even both.
Open source is often DIY :sunglasses:

PR #1649 (2023-07-29): Reimplement sensor statistics in UI
Closes: #1426

This is an alternative version of #1646, which doesn't use the RecyclerView.
commit db1a37f0 (2023-07-29): Reimplement sensor statistics in UI
Closes: #1426
closed 2023-07-29 by dennisguse