OpenTracksApp/OpenTracks #1429: GPS disabled warning but GPS works fine

ae5960e8-a6fc-491f-b252-898ecf59af95, 2022-11-24:
**Describe the bug**
When I'm recording a track, a message popup appears at the bottom of the screen, disappearing, them appearing again and again. It says GPS is disabled but this issue has been happening for a while and my GPS recordings work with no issue. I'm using a Bluetooth sensor for speed and cadence.

**To Reproduce**
1. Start a track
2. The issue occurs, repeating again and again

If applicable:
[image]

**Technical information**
 - Device: Galaxy Note 10
 - OS: IodeOS
 - OpenTracks version: v4.1.2

dennisguse, 2022-11-24:
IIRC this message pops when there was not received GPS location for some time (30s?).
So, the message is slightly misleading.

ae5960e8-a6fc-491f-b252-898ecf59af95, 2023-01-20:
This still happens on the latest version today, GPS works as expected without issue, but the warning pops up every couple of seconds. If this warning could be silenced, I think that would work as a quick fix, as the issue is only the annoying pop-up.

nutpantz, 2023-02-04:
Same issue on Galaxy s7 except it's in w big red rectangle. Flashes on and off repeatedly. But recording of track seems fine

ae5960e8-a6fc-491f-b252-898ecf59af95, 2023-04-03:
The message is repeatedly showing up about every second now after I updated to Android 13.

nutpantz, 2023-04-03:
The gps does not work fine with this bug.
I have tested two  Samsung s7's and a black view 9800. And with the bug there can be over 1500 meter differences between the devices on the latest version from fdroid. When i installed v 3.7 of the app that does not have the bug all 3 devices were within 150 meters for being the exact same.

That is quite a difference. I have tested the new app over ten times and the old app twice.
I will be using the old app and will keep testing if it proves more accurate i will keep to the old version.

dennisguse, 2023-04-04:
@nutpantz Sounds like a new issue; could you create one?

nutpantz, 2023-04-04:
I would not say it is a new issue, just that no one has tested the accuracy with this error. After all without the error the accuracy is very very close between two similar devices

nutpantz, 2023-04-13:
If this fix was in 4.31- 12 gfae on fdroid nightly then the error is still there on my pixel 6

dennisguse, 2023-04-13:
@nutpantz this issue was about an error Message that was shown without the situation actually occuring. 

What you described seems like a different thing.

nutpantz, 2023-04-13:
The error message is still showing on the latest nightly i just downloaded. Exactly the same way it was showing before

dennisguse, 2023-04-17:
@nutpantz created #1532

nutpantz, 2023-04-27:
Sorry, but the latest nightly 4.33 still has this bug
The gps disabled is flashing with gps accuracy of 8m on s7 edge with linageos Android 11

dennisguse, 2023-04-28:
@nutpantz could you create a video showing the problem?

nutpantz, 2023-04-28:
How 
[image]
about a picture

dennisguse, 2023-04-28:
@nutpantz a video shows me the interval at which these messages are appearing. 
At the moment, I don't know where the error might be. 
A video is remote debugging with style ;)

nutpantz, 2023-04-28:
That will be tomorrow then

nutpantz, 2023-05-06:
Latest nightly today

dennisguse, 2023-05-07:
@nutpantz can you upload it anywhere else and share the link?

dennisguse, 2023-05-21:
Some more people have problems.
https://github.com/OpenTracksApp/OpenTracks/pull/1514#issuecomment-1551079499

Any suggestions on how to investigate this?

macias, 2023-06-04:
If I may, there are three things here:

a) IMHO if there is no GPS signal received or GPS is disabled there should be an alarm from OT. The reason being tracing could be important part of work, so if somebody is tracing for a long time, and only after an hour she/he realizes it was for nothing is not too good
b) misleading information -- this is just wording, so it is easy to fix
c) differentiating between GPS disabled and GPS signal lost -- this is quite easy actually. The latter is simple a matter of setting timer, and checking when was the last time GPS was acquired. The former was/is Android event saying there was a switch in GPS provider, so it is a mere subscribing to this event and checking if changed went to `false` or `true`. I've done it in C#/Xamarin using Gingberbread API and it works up to Android 13 :-). But current API could change, my knowledge is rusty

nutpantz, 2023-06-05:
The real question is if open tracks is recording any data while it thinks the gps is disabled. Is it affecting total distance or any other datum?
Maybe a error log could be exported, with all possible data to help diagnose any issue . then users can easily provide as much info as possible, even when users don't have the ability to get Android logs.

dennisguse, 2023-06-11:
Can somebody test #1595?
I rewrote the whole status check and added test coverage for this module.

@nutpantz easy the module that did the status checks is likely buggy (or has hidden features), but that does not affect recording of the data :D

nutpantz, 2023-06-11:
I'm not getting the flashing error on the latest test on two devices

commit d653096a (2023-04-04): Bugfix: GPS disabled warning was showing (repeatedly but incorrectly).

Fixes #1429.
commit bf03de6b (2023-04-07): Bugfix: GPS disabled warning was showing (repeatedly but incorrectly).

Fixes #1429.
commit 60961384 (2023-06-10): Bugfix: GPS disabled warning was shown repeatedly.

Fixes #1429.
commit 86144950 (2023-06-11): Bugfix: GPS disabled warning was shown repeatedly.

Fixes #1429.
closed 2023-06-11 by dennisguse