OpenTracksApp/OpenTracks #1442: Recording heartrate even if sensor gets conected after star of recording

Lucas-Sichardt, 2023-01-04:
Is it possible to save the heartrate into the recorded track even if the heartrate senser (BLE) is not yet connected when the track recording was started?
In my case the sensor is not conecting immediatly and needs up to one minute to sen valid data, but I don't want to wait for this before starting my cycling trip eg.

dennisguse, 2023-01-04:
Actually that should be already the case.
Did you have some trouble with that?

Lucas-Sichardt, 2023-01-05:
Hi,

it is true. For me the connection to the heartrate sensor only works before starting to record in OpenTracks. Even: When I start recording and the sensor becomes afterwards I will get no heartrate reading for the rest of recording. But when I stopp recording and start a new track recording immediatly I get the heartrate just immediatly too.
-- 
Lucas Sichardt
***@***.***
http://lucas.sichardt.net

(Diese E-Mail habe ich von einem Mobiltelefon aus geschickt; eine eventuell erhöhte Häufigkeit von dadurch bedingten Tippfehlern bitte ich zu entschuldigen.)

dennisguse, 2023-01-05:
Mhhh. That's not how it should be.

Which sensor are you using?
Does the sensor behave the same with other applications like FitoTrack?

Lucas-Sichardt, 2023-01-05:
I have a Polar OH1 Model 2L.
It behaves normal with other apps like Forumslader App eg.
-- 
Lucas Sichardt
***@***.***
http://lucas.sichardt.net

(Diese E-Mail habe ich von einem Mobiltelefon aus geschickt; eine eventuell erhöhte Häufigkeit von dadurch bedingten Tippfehlern bitte ich zu entschuldigen.)

dennisguse, 2023-01-05:
Ok.

So, this doesn't work:
1. Start a recording with OpenTracks
2. Turn on Polar OH1
-> OpenTracks doesn't get heartrate

And this works:
1. Turn on Polar OH1
2. Start a recording with OpenTracks
-> OpenTracks gets heartrate from the sensor

Right?

Lucas-Sichardt, 2023-01-05:
Absolutely, that is right.
-- 
Lucas Sichardt
***@***.***
http://lucas.sichardt.net

(Diese E-Mail habe ich von einem Mobiltelefon aus geschickt; eine eventuell erhöhte Häufigkeit von dadurch bedingten Tippfehlern bitte ich zu entschuldigen.)

Lucas-Sichardt, 2023-01-07:
> Well well well.
> That was an ancient bug :)
> On my phone, OpenTracks tries to connect for 30s and then it gives
> up.
> If the connection is lost: it gives up...
> The solution is to trigger a re-connect, whenever a disconnect
> happens (can also be a never connected timeout).
> I tested with a Polar OH1 (because it can be turned on and off).
> The fix will be in tomorrows nightly.

Thanks for your fast response.
I updated the application last night and today I had the chance to test
it on bike rides. For me it still doesn't work.
Did I not get the needed version, perhaps?
It's OpenTracks v4.2.0/4.2.0 5119...
I had the repository for nightlies added via the QR-Code in Github...

Lucas-Sichardt, 2023-01-08:
> That's the last release (not containing the change), which is
> published by F-Droid. If added the nightly repo to your F-Droid,
> OpenTracks will be shown twice: the release and the nightly.
> Also the nightly has a black launcher icon.

Hi, 
thanks for the information. I now installed the nightly-version as
well. But unfortunately the problem is still the same...
(version name v4.2.0-4-g12b3d31f5 (code 5124)

dennisguse, 2023-01-08:
Mhh.
Then I need some more info.
Are you able to create a logcat with the current nightly?
And which Android version / device are you using?

Lucas-Sichardt, 2023-01-08:
I am using a BQ Aquaris V and the Android Version is 8.1.0

How can I find that log?
-- 
Lucas Sichardt
***@***.***
http://lucas.sichardt.net

(Diese E-Mail habe ich von einem Mobiltelefon aus geschickt; eine eventuell erhöhte Häufigkeit von dadurch bedingten Tippfehlern bitte ich zu entschuldigen.)

dennisguse, 2023-01-09:
Simplest is this: https://developer.android.com/studio/command-line/logcat

Lucas-Sichardt, 2023-01-11:
Hi again,

unfortunately I did not have the time yet to figure that out. I have been using ade-features in the past, but it's been a while since then.

But today I had another thought while wondering why your solution should not work for me and I just tested. Perhaps I found the "bug" or feature:

The heartrate is shown and recorded even if the sensor is switched on after starting recording in OpenTracks - but only if Bluetooth was switched on in the moment when OpenTracks recording is started.

So my problem is, that I have Bluetooth switched off normally and I have an application that switches it on when OpenTracks starts recording (recognized by the notification at the top of the screen).
And when I was testing manually I also switched Bluetooth on and off as intented...

Best regards,

Lucas
-- 
Lucas Sichardt
***@***.***
http://lucas.sichardt.net

(Diese E-Mail habe ich von einem Mobiltelefon aus geschickt; eine eventuell erhöhte Häufigkeit von dadurch bedingten Tippfehlern bitte ich zu entschuldigen.)

commit 12b3d31f (2023-01-05): Bugfix: BluetoothLE requires to trigger reconnect on either disconnect or never connected.
Initial connection timeout seems to be 30s.
Fixes #1442.